ShikOfTheRa/scarab-osd #381: Feature suggestion/request: wind direction & speed indicator (calculation proposal for fixed wing)

Dronek, 2017-02-16:
It would be nice to have wind direction and speed indicated by the OSD similar to RTH arrow. For wind, the arrow could indicate the wind direction relative to the aircraft and the value below could indicate wind speed in a unit of choice (ideally identical to other speed indicators' units on the screen, i.e. kph).

I suggest calculating wind speed and direction during a segment of leveled and straight flight based on the following sensor data:

- MAG heading
- airspeed
- GPS heading
- ground speed (GPS based)

I sketched the vector-based calculation below. However, not 100% sure if it's correct.
Discussion appreciated :)

[image]

u8mydog, 2017-02-16:
I'm very rusty at this stuff but I did create comparable calculations many many years ago when I created a 'basic' program on a pocket TRS-80 to simplify creating flight plans when I was a current private pilot.
Luckily you are not trying to calculate climb and decent rates when taking into consideration air density.
But, I question why you would want to add these calculations/filtering to an OSD controller because these little devices are memory limited.
My suggestion would be to use LUA scripts on an OpenTx radio such as Taranus or Horus. The radio would receive the sensor info via telemetry, calculate wind speed/direction, and then send the info back to the FC either as telemetry info, or via dedicated channels (much like how RSSI can be displayed on OSD).
Then the LUA could be adopted to collect wind speed/direction from a ground station that can send the info to the TX via the training port.

Dronek, 2017-02-16:
> But, I question why you would want to add these calculations/filtering to an OSD controller because these little devices are memory limited.
My suggestion would be to use LUA scripts on an OpenTx radio such as Taranus or Horus. The radio would receive the sensor info via telemetry, calculate wind speed/direction, and then send the info back to the FC either as telemetry info, or via dedicated channels (much like how RSSI can be displayed on OSD).
Then the LUA could be adopted to collect wind speed/direction from a ground station that can send the info to the TX via the training port.

Oh thanks for your feedback. I had no idea about how busy OSD processors are, doing their OSD job. Lua Script is a nice idea to source out the "heavy lifting". It's just that I don't have any of the available channels free to send this data back to the plane. Also, with crossfire CRSF telemetry the update rate is currently 1Hz which is really slow.

ShikOfTheRa, 2017-02-17:
Plenty of power available. Not a lot of memory though:)
This is possible. I have added to the list of development requests. Unfortunately I have a lot on mine - I hope someone else can pick up 

Where do you get the airspeed? There is code in OSD for this, but it hasn't been tested or used as far as I am aware.

Dronek, 2017-02-17:
> Plenty of power available. Not a lot of memory though:)
 This is possible. I have added to the list of development requests.

Nice, thanks :)

> Where do you get the airspeed? There is code in OSD for this, but it hasn't been tested or used as far as I am aware.

I'm using iNav on my flying wing and as far as I know airspeed sensor support is on the way in iNav.
Being no coder, I guess it shouldn't be a big issue to get airspeed from iNav as soon as it's available?

ShikOfTheRa, 2017-02-27:
@dronek - I am going to close this ticket for now to keep the issue list clean.
It is on the wishlist for next release. Too late for this next one.
Wake it up once iNav has  a version released where airspeed is available as an fyi.

closed 2017-02-27 by ShikOfTheRa

Dronek, 2017-10-26:
> Wake it up once iNav has a version released where airspeed is available as an fyi.

Meanwhile iNav is able to log raw data from an airspeed sensor. I don't know if this helps but maybe MWOSD could make use of this. Sorry, I see no option how to open this issue...