ShikOfTheRa/scarab-osd #387: Sidebars scrolling another direction

EvgMe, 2017-02-23:
Sidebars must scroll as real HUD: When value increase, sidebar must scroll down. And when value decrease, sidebar must  scroll up.

ShikOfTheRa, 2017-02-24:
I don;t think anyone has noticed before. Maybe no real just pilots fly FPV!
For both speed and altitude?
I can easily add a choice.

EvgMe, 2017-02-24:
Yes, both spead and altitude. For example it's shows for mission planner: https://www.youtube.com/watch?v=XCSgy9_XFKo. Or B737-800 https://www.youtube.com/watch?v=IO_LJd-iO9o
 The choice is a good idea.

ShikOfTheRa, 2017-02-24:
OK - will implement choice option..

In future release I hope to try some live font changes to create an effect similar to the hud on the video. Its difficult with current hardware and impelmentations.

I guess you are using APM? How do you find it. I don't think many use MWOSD with APM/PIXHAWK.

ShikOfTheRa, 2017-02-24:
Should be next next week... Very simple change

macfly1202, 2017-02-24:
there a good implentation of this hud in playuav osd. I made some test in the past and working great (but it was a stm32F1) with a pixfalcon (stm32f4). Here an example https://youtu.be/K_rhIYWg0L0 or https://youtu.be/J-fHLHofqTc 
. The last active fork of playuav is here : https://github.com/TobiasBales/PlayuavOSD

ShikOfTheRa, 2017-02-27:
Closed / done

I have amended and uploaded. Its first request for change and the scrolling is eyecandy so decided not to make it a choice. It is fixed in correct direction now.

I am working on an stm based OSD for myself that isn't limited by the max graphics. I try to do in color but its a lot more difficult with timing and don't have the time anymore so will stay with black/white  
I forgot about playuave - might have to look for some ideas.

closed 2017-02-27 by ShikOfTheRa