ShikOfTheRa/scarab-osd #388: incorrect max distance in summary

csurf, 2017-03-03:
I just had a flight with most recent version from repo, and discovered that the 'max distance' value displyed in the flight summary was way off. I had a flight with a max distance of 4.2km/4225m, and the summary page showed a value of '2359.2' (also seems strange that it's shown as a float value). All of the other values on the summary screen looked ok, but the max distance was definitely off by a lot.

I can provide screenshots if necessary.

ShikOfTheRa, 2017-03-03:
Just tried on a simulator - got 12.9 total distance with 4.465 meters max distance - displays OK.

max distance is the highest value it thinks it has received from the FC. 
with fixedwing, if it sees >9999m it reports in km. So thats 2359.km away. A new FPV record!!!
FC sends this as an actual value. Its not calculated by MWOSD.
So could be something received from the FC or MWOSD itself. 
Most likely a GPS glitch or corruption in data ( I see plenty in testing!)a wobbly. 

Had a look at code and it looks simple. Not much to go wrong.. 
Hard to say for certainty, but my money is on GPS data corruption...

I think best thing I can do is put a glitch filter to limit the change to say 100m change per update to account for bad data. OK with that?

ShikOfTheRa, 2017-03-03:
Tested and uploaded... Should stop any crazy values.

macfly1202, 2017-03-03:
I would like to add (but never tested mwosd with fixed wing but only ardupilot) only take value from gps if there is a 3D/DGPS fix OR hdop < 10meters for example ...all clever autopilot use this kind of triger....

csurf, 2017-03-03:
@ShikOfTheRa 
ok, will try to test today. 
If it was due to a random glitch, will be hard to know if it really went away. However, your theory makes sense, so hopefully the glitch filter will take care of it. FYI, this is with latest iNav 1.6.

ShikOfTheRa, 2017-03-06:
@cssurf - I did some testing. With a continuous big glitch is took 15 secs and many updates to catch up the glitch distance. So his looks like it works. 

@macfly1202 - in this the data comes from the FC as trusted. Due to amount of memory available there, data validation and menu configurable limits have very little data validation. All problematic ones done a long time ago. Also hdop not normally available to OSD - that's down to the FC. That said, in testing for the GPSOSD variant, I did find that data from GPS was variable. Sometimes good sat / hdop, but bad data. This may have been corruption or possibly during the GPS sending it changed from good to bad in that fraction of time. Not sure really, but definitely the odd glitch.

ShikOfTheRa, 2017-03-06:
CLosing for now. Feel free to continuing comments - or re-open if required

csurf, 2017-03-08:
@ShikOfTheRa,
ok, so I had two flights today, both over 4km, ran into the issue again. I got the same max distance value of '2359.2' at the end of each flight.

Could this possibly an iNav issue? Where exactly does the max distance value come from? I don't know how else to test this. 

I can post a video of one of my flights if it'll help, but there's not much going on, just a long, boring fixed-wing 4km flight out&back. I didn't have any issues during the flight (failsafe, GPS glitches, etc.). Any ideas?

ShikOfTheRa, 2017-03-08:
Yeah that's weird. If you can post the vid I'll take a look. Not boring to me - sad that it is I do watch vid's on YT with MWOSD in them!

No idea on the cause - haven't been able to replicate. If can't see any hint on the vid,might have to give you a build with a couple of debug values. to see if its mwosd or soemthing coming from inav. I have used patrike witha wing over those distances but not inav yet. Got a plane ready for maiden.

csurf, 2017-03-08:
@ShikOfTheRa,
ok, here's the vid.
https://www.youtube.com/watch?v=FnKWpLFPoJE
please let me know if you need anything else... I really would like to get this fixed (trying to set some personal distance records on a couple of planes, and it helps to have the 'max dist' logged correctly )

csurf, 2017-03-08:
@ShikOfTheRa,
FYI, forgot to mention that I have long-range display mode enabled (I have a feeling that the bug may have to do with that). I'm going to disable it, re-flash, and see if it happens again tomorrow. Will report back. Lemme know if u find anything else...

ShikOfTheRa, 2017-03-08:
It should auto enable for fixed wing.
Nice to see the vid... Can't see anything obvious in it ...

Uploaded a version which has this -
/********************  RECORD CAPTURE  settings      *********************/
//#define SHOW_MAX_DISTANCE              // Enable to display MAX distance achieved on line below current distance

Lets see live how and when it displays a weird number.
I have also made some changes that may be related to the issue - although I can't replicate.

On a positive - did find a bug in another part - max altitude display!!

[image]

csurf, 2017-03-09:
@ShikOfTheRa,
ok, i took a couple of test flights today. After reviewing the DVR vids, I found that the problem happens at startup (perhaps when home position gets set?). The first flight displayed the problem, the second flight didn't. The first flight showed '2359.2' from the point when I started recording. Home was already set & it was prior to arming the plane. The 'max distance' display showed a value of '2359.2' for the entire flight even after having flown out past 3-4km. 

Unfortunately, there was a small range of time between when I powered up the plane & when I hit 'record' on DVR (I was getting my gear ready), so I don't know exactly at what point the value went from '0' to '2359.2'.

I will try to test on bench to see if it happens again.

Could this have something to do with how the distance is read from FC?

On the second flight, everything worked normally. Unfortunately, I didn't go too far, so max distance didn't get pushed into the 'long range display' range (only 700m).

csurf, 2017-03-09:
@ShikOfTheRa,
ok, bench tested but couldn't reproduce yet.

I guess perhaps iNav is sending some junk distance data before it gets a good 3D GPS lock? One option could be to reset the distance on first arm?

also noticed there's a lot of sign/type changes for the distance values (uint16, uint32, int32). Not sure if that might be causing some issues somewhere if it gets out-of-range data somehow...

csurf, 2017-03-09:
@ShikOfTheRa,
ok, think I figured out how to reproduce it, and it seems like a pretty nasty bug.

I modified the code to print both the raw distanceMAX value (uisng the itoa() function) & and the formatted value that's returned from the formatDistance() function. 

  #if defined SHOW_MAX_DISTANCE
    formatDistance(distanceMAX,1,2);
    MAX7456_WriteString(screenBuffer,LINE01+2);//getPosition(GPS_distanceToHomePosition));
	
    itoa(distanceMAX, screenBuffer, 10);
    MAX7456_WriteString(screenBuffer,LINE01+12);//getPosition(GPS_distanceToHomePosition));
  #endif //SHOW_MAX_DISTANCE

Everything works fine *as long as* the plane isn't is rolled past around 30 degrees. After that, the formatDistance() function begins to fail for some reason, and it starts printing that crazy 2359 number. Also, the value printed by itoa() still looks valid, but it actually appears to be stuck.

Here are some example videos:
https://www.youtube.com/playlist?list=PLrdiKwHGRZDsFluka3oyFL07_up68qUPH

csurf, 2017-03-09:
ok, found the source of the bug, but not the exact code.
It's caused by the AHI function, displayHorizon()
I disabled the code in that function by return immediately from it, and the distance bug goes away. Enabling the AHI code causes the bug to return.

I also tried disabling the pitch & roll code, thinking maybe this was the cause, but the bug was still there, so it's further down in the AHI code.

This is as much as I can work on it for now. Hopefully, you'll be able to pinpoint the source of the bug given this info. :)

csurf, 2017-03-09:
@ShikOfTheRa 
think I found it, the HB elevation code...
Can you take a look at it? it must be overflowing or something...

ShikOfTheRa, 2017-03-09:
Awesome. I will check it out today.... Thank you for all the testing. Being able to replicate and narrow it down really helps...

ShikOfTheRa, 2017-03-09:
OK - I think with sidebars enabled the screen array is overflowing. There are no checks for this.
Added some in... Can you try?
Also added in the #define SHOW_MAX_DISTANCE into config.h I got dropped in my upload

I couldn't replicate - possibly down to different compiler / options enabled, but definitely an issue...

csurf, 2017-03-09:
@ShikOfTheRa 
ok, the bounds checking was the problem...
I ended up doing the following:
```
        if (Settings[S_HORIZON_ELEVATION]){ 
          if(X >= 2 && X <= 6) {
			int16_t p = (pos-(3*LINE));
			if(p>0){
				screen[p] = SYM_AH_BAR9_0+(Y%9);
			}			
			p = (pos+(3*LINE));
			if(p<480){
				screen[p] = SYM_AH_BAR9_0+(Y%9);				
			}		
          }
```
that'll save us a couple of extra redundant calculations. Is that cool?
Anyway, problem appears to be fixed, will flight test tomorrow if possible & post back...

csurf, 2017-03-09:
side-note:
I think the horizon/AHI stuff needs somewhat of an overhaul. It should be able to work with the plane inverted (maybe with some kind of indicator showing that its upside-down/rolled over 180 degrees). 

Also, after applying the patch, I briefly saw the altitude glitch to that same '2359' value, before going back to zero. I tried to reproduce it, but haven't been able to. Hopefully it won't happen at some critical point during flight. 

There definitely appears to be some general memory/overflow issues that might need a closer look in general.

Anyways, I will try to test  & report back later today or tomorrow. Thanks again...

ShikOfTheRa, 2017-03-09:
Yeah there is some significant opportunity to improve and reduce memory. Its grown somewhat in pieces over time. Just having time to do it properly... Maybe next release will concentrate on code improvements. Most of all I want to reduce memory, more items to GUI and flash from GUI

Code suggestion looks good. 
Also needs also to be in normal elevation part and full ahi because someone may put high values into AHI max in config.h.

I am tied up with work today so only been able to have a quick look first thing

ShikOfTheRa, 2017-03-10:
Back in the office today so can have a proper look and bit more time to respond.

If you use your method, you'll also have to do this or some variant to avoid overflow:
        uint16_t pos = position -9 + LINE*(Y/9) + 3 - 4*LINE + X;
        if ( pos >0 &&  pos < 480) // add thsi check
          screen[pos] = SYM_AH_BAR9_0+(Y%9); 

Also, using a uint instead of int has advantage of only having to check for >480 because any negative values become large positive..
Looking back in history, no constrain was required because of the values used. 
Now its adjustable parameter and they have increased we need the checks on both horizon + elevation bars.

An area to revisit after 1.7 is out. Interesting to see how it will worked flipped etc. 
As fyi, I recall someone else having problem with AHI display quite a while back. I wonder if this is related.
]
Good catch csurf!

csurf, 2017-03-13:
@ShikOfTheRa 
- ok, point taken on the 1st test.
For clarity, we could probably just do a constrain() right off the bat, combining the math ops into a more compact form...
```
uint16_t pos = constrain( (position - 6 + X + LINE * ( (Y/9) - 4)), 0 ,480);
//fyi, math ops above are combined
```
cool?

- Good point on using the uint & avoid the 'less than zero' test. 
In any case, we'd still need to do ' p < 480'  twice, once for each operation, but it'll work fine...
```
        if (Settings[S_HORIZON_ELEVATION]){ 
          if(X >= 2 && X <= 6) {
			uint16_t p = (pos-(3*LINE));
			if(p<480){
				screen[p] = SYM_AH_BAR9_0+(Y%9);
			}			
			p = (pos+(3*LINE));
			if(p<480){
				screen[p] = SYM_AH_BAR9_0+(Y%9);				
			}		
          }
```

how does all of this look to you?

FYI, I managed to get the AHI to work in the inverted position, but there are a couple of improvements that I thought might be cool to add to it. I'm moving this to separate issue...

ShikOfTheRa, 2017-03-14:
For this, remember the constrain needs to be later. Because could add at later stage.

An alternative would be a small function like below. Also saves 20 bytes of memory.

void constrainedScreen(uint16_t pos, uint16_t Y) {
  if (pos<480)
    screen[pos] = SYM_AH_BAR9_0+(Y%9);
}

and:
    for(uint8_t X=0; X<=12; X++) {
      if (X==6) X=7;
      int Y = (rollAngle * (4-X)) / 64;
      Y -= pitchAngle / 8;
      Y += 41;
      uint16_t pos = position -9 + LINE*(Y/9) + 3 - 4*LINE + X;
      constrainedScreen(pos,Y);
      if (Settings[S_HORIZON_ELEVATION]){ 
        if(X >= 4 && X <= 8) {
          constrainedScreen(pos-3*LINE,Y);
          constrainedScreen(pos+3*LINE,Y);
        }
      }
    }

In truth The whole function could probably be optimised. Looks some  duplication. Possible using X from -4 to +4

ShikOfTheRa, 2017-03-22:
Going to close this as resolved now.

I have added into development roadmap to revisit the AHI section to improve memory / performance. Definitely opportunity.

Thanks for input and testing - brill.
Will update download  files too.

closed 2017-03-22 by ShikOfTheRa