ShikOfTheRa/scarab-osd #391: move GUI code to separate repo

csurf, 2017-03-05:
I realize that the old GUI is kinda junky & overloaded, and will probably soon be replaced by the chrome app, but, in the meantime, how about sending it to its own, separate repo? Also, it would help to remove all of the platform build stuff (from processing app) & simply have the code & data files placed in the repo. Would make it much easier to work with the GUI code, and would also make it easier & quicker for users to clone the the MWOSD repo (wouldn't have to deal with a massive download due to GUI stuff). Sound reasonable?

ShikOfTheRa, 2017-03-05:
Agree. Was heading that way soon. Now is as good a time as any. I want to repackage 1.6 with new gui as well. 
Will do this week

ShikOfTheRa, 2017-03-06:
It is done:

https://github.com/ShikOfTheRa/scarab-osd-gui

I think its at end of its life. 
I'm not planning to keep it in updated anymore. Haven't used it for some months now.

closed 2017-03-06 by ShikOfTheRa