ShikOfTheRa/scarab-osd #393: Allow MSP sending voltage and amps to FC

sppnk, 2017-03-06:
As MWOSD can take variables from FC like RSSI and others, why dont allow the reverse thing? Could be done? (of  course FC should to be able to take them).

This is because in many setups the volts and amps are only in the OSD, but sometimes we fly the same airplanes w/o vtx, third view, and would be nice the FC to have this (for telemetry at least).

I will open the same request in INAV repo.

What do you think?

ShikOfTheRa, 2017-03-06:
Can be done - if can convince iNAv guys. 
Think it will be low on their list as could connect direct to the FC in most cases?

sppnk, 2017-03-21:
Hi, I close this as I am taking the values directly in the FC. I guess it is better and I dont know if the effort to modify MSP worths this. Closing...

closed 2017-03-21 by sppnk