ShikOfTheRa/scarab-osd #397: Betaflight OSD Menu From Transmitter

jcutcher01, 2017-03-09:
When powered on and bound, I yaw left pitch forward and the OSD menu does not appear. The motors beep twice like when the FC resets. When I go into the configurator using the chrome app, I see my PID profile changed from #1 to #2. 

Now I was not center throttle so I can try that later when I get home. Someone also mentioned try yaw right. 

I am running the Flip32 F4 Ominbus for the first time and I have 3.1.6 loaded.

Also, in the configurator OSD menu using Chrome App on a MAC, I cannot scroll all the way down to see all of the selectable settings.

closed 2017-03-10 by jcutcher01

jt6064, 2017-07-12:
What was the fix for this?  I'm experiencing the same problem on a new Betaflight F3 board.  Stick commands are not working.  I've even tried the throttle in different positions.

jcutcher01, 2017-07-12:
My issue was I forgot to set the throttle to mid plus Yaw left. I had throttle all the way down and I was yawning left.

jt6064, 2017-07-12:
Ok, sir.  Thank you very much.  This may seem silly but did you have the copter armed?  I know that no throttle signal is sent when the copter is disarmed via a switch.  The switch only arms the quad if the throttle is in low position.  Are you raising the throttle to mid position and flipping the arming switch?

jcutcher01, 2017-07-12:
Quad was disarmed. Mid throttle left Yaw pitch forward works for me.

jt6064, 2017-07-12:
Thanks again.  One last question....any explanation ever given for why the stick command changed your PID profile?

jcutcher01, 2017-07-12:
No. I did observe this on my Kombini FC so I wonder if BF monitors sticks when unarmed for switching profiles?