ShikOfTheRa/scarab-osd #398: Closed source releases

alvarofrank, 2017-03-12:
After Joshua Bardwell's video about GPL, I am wondering if OSDs like the FuriousFPV Passado OSD VTX are violating the MWOSD license.
They provide a binary preflashed for the Passado OSD without the source code.
Upon request they refuse to provide the source:

> Alvaro Frank:
> Mar 9 (3 days ago)
> 	
> to TECH
> Hi Bas,
> 
> Thank you for your reply.
> Where is this costum own 1.6 version available for download?
> TECH FURIOUS FPV
> 	
> Mar 9 (3 days ago)
> 	
> to me
> Hello Alvaro,
> 
> I don't have that, they have it on the factory, but we don't release that version for public.
> 
> Best Regards Bas

I will try to bring this to their attention again and perhaps get the source, since I would like my purchased hardware to use the newest versions of MWOSD.
Any opinions?

DieHertz, 2017-03-12:
If they use MSP, they clearly violate GPL.

digitalentity, 2017-03-13:
@DieHertz technically they can re-implement MSP from specs. GPL only protects implementation (code), not specs. However it seems more likely that they simply took the code.

ShikOfTheRa, 2017-03-13:
Agh - these guys make some great stuff but its like pulling teeth getting the software.
@alvarofrank can you request that they release it under the GPL terms associated. Advise them it is a legal (and moral) obligation.

My company makes more from copyright litigation than its products. Its unfortunate, but it is now considered a business revenue stream....  
They would be wise to comply because a number of businesses have been closed due to cost of fine + crazy legal costs.

alvarofrank, 2017-03-13:
Thanks for the feedback, I am just confused, because the software is literally a modified 1.6 version. Regardless of MSP, MW/Scarab OSD should force their custom version under GPL right?
I send another message last week and am waiting for reply. 
The problem is that currently the OSD resets randomly to unusable default settings.

ShikOfTheRa, 2017-03-13:
I believe it is standard 1.6 too.

I will enforce GPL- give them an opportunity to produce first. They did for the Innova and Piggy - Eventually!

But note there is no support for it under normal MWOSD. There is a VTX development branch  which hopefully be integrated at some point.

ShikOfTheRa, 2017-03-22:
I have requested...

ShikOfTheRa, 2017-04-05:
Requested again

ShikOfTheRa, 2017-12-07:
Under duress I have passed details of the infringement to GPL enforcement. I don't want anything else to do with it so closing this.
It's not right, but personally I don't really care about it to much as its an older version with less features and more bugs. 
Am sure they are getting enough grief from their user base :)

closed 2017-12-07 by ShikOfTheRa