ShikOfTheRa/scarab-osd #400: make AHI work for inverted flight

csurf, 2017-03-13:
I've got the following fix in order to get the AHI to correctly show when the aircraft is inverted...
in screen.ino, displayHorizon() function:
```
  if(pitchAngle>AHIPITCHMAX) pitchAngle=AHIPITCHMAX;
  if(pitchAngle<-AHIPITCHMAX) pitchAngle=-AHIPITCHMAX;

  if(rollAngle > 900){
    rollAngle -= 1800;
  }
  if(rollAngle < -900){
    rollAngle += 1800;
  }
  if(rollAngle>AHIROLLMAX) rollAngle=AHIROLLMAX;
  if(rollAngle<-AHIROLLMAX) rollAngle=-AHIROLLMAX;
```

This seems to work fine. The only problem is that it fools the subsequent code into thinking that the plane is upright. Ideally, you'd want to adjust the AHI to somehow indicate that the plane is upside down. 

I can't recall where, but I remember seeing an AHI where some of the markers in the horizon bar would shift out of position whenever the plane was upside down. Something similar to the following...
```
   -  -  -  -  -  - 
- _ - _  *  _ - _ -
    -  -  -  -  -  -
```
not sure if that makes sense, but it basically draws alternating 'low' lines in the horizon bar. Perhaps I'm miss-remembering, but that's how I *think* I remember seeing it work in some other AHI setup (not sure if it was from an FPV or a real-world HUD).

ShikOfTheRa, 2017-03-14:
Cool. TBH, if I'm upside down I aint looking at the AHI - I'm usually crashing or enjoying the view!!! Love doing that in my funjet. 

It would be good if can manipulate it to work correctly.
Do need an indicator? I don't recall ever needing an AHI to tell me I'm upside down. Memory is tight and prepared to give up bits that make it perfect. Or add it as a #define option for those that want it.

maybeyes, 2017-05-15:
Nice feature ;) I need it.

closed 2018-01-11 by ShikOfTheRa