ShikOfTheRa/scarab-osd #401: R1.7  not conecting GUI in mode Standalone-OSD.

isabido, 2017-03-19:
Hi guys, I'm using version 1.7, to use standalone OSD with GPS.

I followed this guide:
Http://intofpv.com/t-osd-standalone-gps

If I enable the option "#define GPSOSD_UBLOX" I can not then access it with the GUI for its configuration.

Any suggestions?

Thank you!

ShikOfTheRa, 2017-03-19:
1 - are you using same baudrate on OSD and GUI
2 - GPS must be disconnected when connecting to GUI

isabido, 2017-03-19:
1- Yes, the baudrate is the same in the GUI.
#define BAUDRATE 115200.

2- The GPS is obviously disconnected.

I have already commented that just disabling the UBLOX support works correctly.

ShikOfTheRa, 2017-03-19:
Ok thanks. I will check first thing in morning....

ShikOfTheRa, 2017-03-20:
Bug confirmed. Will get this fixed ASAP.

isabido, 2017-03-20:
Thanks for your confirmation, meanwhile version 1.6 is not affected?

Regards!

ShikOfTheRa, 2017-03-20:
I think 1.6 will be same. I think I understand the issue. Trying some options today...

ShikOfTheRa, 2017-03-20:
OK - made amendment to OSD code. I am unable to test with GPS chip at moment to confirm nothing is impacted, but it looks OK on simulator...! Please try.

https://github.com/ShikOfTheRa/scarab-osd/archive/master.zip

ShikOfTheRa, 2017-03-22:
Confirmed by others this is now working too

closed 2017-03-22 by ShikOfTheRa

isabido, 2017-04-01:
Confirmed, works correctly. I have opened a new issue. Does not display correctly with "arrows" the address to home. Instead it shows "LETTERS"

I try to load the fonts, and it disconnects when it reaches 10%.