ShikOfTheRa/scarab-osd #402: Stupid Question: How do I upload the config.h file to the OSD?

SirPotato, 2017-03-21:
I'm running a Diatone SP3 VTX/OSD combo, and I can't figure out for the life of me how to get the the config.h file to be loaded onto the OSD. I thought just saving it would work, but it didn't. I can configure it through the GUI, but I want some of the settings in the config file that are not in the GUI to be activated. I'm new to OSD configurating so this is probably a stupid question. Thanks in advance!

ShikOfTheRa, 2017-03-21:
Here is a very basic step by step guide. Don't miss a single line:
https://github.com/ShikOfTheRa/scarab-osd/wiki/Quick-start-guide

In brief you need to install arduino, edit config.h flie in that and upload to the board.

SirPotato, 2017-03-21:
Ah, thank you. I had tried opening it in arduino, but I had tried to open config.h directly instead of the .ino file.

closed 2017-03-22 by ShikOfTheRa