ShikOfTheRa/scarab-osd #404: course over ground with INAV

sppnk, 2017-03-29:
Hi @ShikOfTheRa . The 'arrow', as direction to home, is working with INAV and LTM? I have found that it goes crazy, when using LTM telemetry (I share the OSD UART with telemetry device)

In LTM, no 'course over ground' is sent (only attitude heading in 'A' frame), and I cannot find in your code the calculation of this value and if it is compared to home heading in order to draw the correct 'arrow' direction or show the value in degrees. 

I am playing lately with a DIY LTM reader, and I have the same problem, so what I will try to do is to calculate myself that 'course' value taking different GPS points of the actual track.

Am I wrong with this? Thanks.

ShikOfTheRa, 2017-03-29:
Firstly, I am not sure all FC code use the same meanings for direction... 
but for iNav and and LTM, my assumption has been they use the same data as for MSP

It gets heading to home from G frame:      GPS_distance_cm_bearing(&GPS_latitude,&GPS_longitude,&GPS_home[LAT],&GPS_home[LON],&dist,&dir);
      GPS_directionToHome = dir/100;

To this it applies correction from actual heading/cog (default=mag on multi, GPS on fixedwing) 

mag heading comes from  A frame
gps heading comes from magic... er it looks like I forgot to generate GPS heading in LTM mode for fixedwing... bug alert!

Can force it by #define USEMAGHEADING, but if no mag you will probably get crap data. If do have a mag they have to be away from power wires.  

So to fix:
1 - we ask iNav devs to send gps cog instead of mag heading if no mag fitted. Remove fixedwing USEGPSHEADING from LTM
2 - we save lat/lon each frame and if deifferent to previous, calculate new cog using GPS_distance_cm_bearing

sppnk, 2017-03-29:
I use mainly cc3d for planes, so.. no MAG, hehe.

I like second option, as I guess LTM is not easy to change, because it is growing the use of it in different systems .

ShikOfTheRa, 2017-03-29:
Well its not really changing LTM protocol - just sending valid data !

At last I am using iNav on a plane now!!. I picked a funjet - probably not the ideal first plane to learn iNav with!! Had its maiden flight. It went really well. Need to do some tuning though

https://www.youtube.com/watch?v=U3tDgGzDvkc

sppnk, 2017-03-29:
Great !!. Nice flight. I have a Miini Talon with INAV and high speeds. It works fine.

This issue is treated here in INAV repo: https://github.com/iNavFlight/inav/issues/1416

Dronek, 2017-03-30:
Slightly related to [this](https://github.com/ShikOfTheRa/scarab-osd/issues/381): 
would it be feasible to display MAG heading and GPS heading (course over ground) together in the compass indicator?
For example: 
 |   |   S   |   |   W   X   |      <---- MAG heading with "X" indicating GPS heading

Would be helpful for the pilot to estimate drifting due to crosswind.

ShikOfTheRa, 2017-03-30:
Added bothe to development roadmap / wishlist:

https://github.com/ShikOfTheRa/scarab-osd/wiki/Features-future-roadmap

sppnk, 2017-04-02:
Just to say I have tested again after clean some bugs in my LTM reader code, and now I see that 'heading' value from LTM is perfectly the 'course over ground' when plane comes home in RTH.

I will test in next weeks the osd arrow issue. Thks.

sppnk, 2017-04-03:
I will close this until new tests. Thks and sorry for the mess.

ShikOfTheRa, 2017-04-05:
Looking at DE's response I think I will remove the use gps component when fixedwing defined.

Until then, FOR LTM ONLY... use #define USEMAGHEADING (even though no MAG)

sppnk, 2017-04-07:
Hello @ShikOfTheRa , I reopen this becasue I tested in flight yesterday and I found that the arrow showing direction to home doesnt work. It shows wrong direction.

As at the same time I am cheching this with my LTM reader, I can say that the data coming from INAV telemetry LTM ('A' frame) or the computation I do at the same time are correct. 

But the arrow doesnt. Could be a missconfiguration or a bug? Thanks.

ShikOfTheRa, 2017-04-07:
There would have been an issue if using FIXEDWING and LTM on any builds prior to updates to the repository yesterday.

If its still same I'll have to try to throw some real data at it as I can't see any more errors from just looking at the code.

sppnk, 2017-04-07:
Ok, I have not actualized yet. I will try this new version and report. Thanks.

ShikOfTheRa, 2017-06-13:
Assumke all OK now. Please re-open if issues

closed 2017-06-13 by ShikOfTheRa