ShikOfTheRa/scarab-osd #406: Failed to upload fonts.

isabido, 2017-04-01:
Does not display correctly with "arrows" the address to home. Instead it shows "LETTERS"

I try to load the fonts, and it disconnects when it reaches 10%.

This is the version used.

https://github.com/ShikOfTheRa/scarab-osd/archive/master.zip

ShikOfTheRa, 2017-04-03:
Can you advise what changes you make to config.h?

isabido, 2017-04-03:
Something like that...

http://intofpv.com/t-osd-standalone-gps
#define MINIMOSD 
#define GPSOSD_UBLOX 
#define FIXEDWING
#define BAUDRATE 115200

Thanks!!

ShikOfTheRa, 2017-04-03:
Ah .. ok. I have an idea.

ShikOfTheRa, 2017-04-04:
Bug confirmed. Thanks for reporting. 

This would impact UBLOX and MTK users. Very few which is possibly why not noticed before.
Its actually a bug in chrome that OSD is trying to workaround. Unfortunate side effect I did not think about..

Fixed and  uploaded into github. Please test and advise if OK.

isabido, 2017-04-04:
Thank you for your prompt response, I will not be able to confirm until next weekend.

I'm sorry for the delay, but for work matters I'm not possible.

ShikOfTheRa, 2017-04-04:
OK - no hurry.
I could replicate and the update worked for me. Hopefully will do for you to

closed 2017-06-13 by ShikOfTheRa

eephyne, 2017-07-13:
Hello, sorry to ask on a closed issue but did you add the fix in the last zip 1.7 ?
I have the exact same issue (standalone and UBLOX) and The version I use is the zip dowloaded on the website.
I do not mind using the git repo but wanted to check first if that was useful.

eephyne, 2017-07-14:
I did not test the osd but the upload of the font was succesful (ftdi keep blinking till the end and gui not disconnected.
Thanks @ShikOfTheRa for your time and great software

ShikOfTheRa, 2017-07-14:
Great! glad its working OK