ShikOfTheRa/scarab-osd #408: Question

bfasty, 2017-04-13:
I'll start with  I am not a coder.  My question is... would it be possible to make a version of this OSD that does NOT need FC communication?  I know you would loose certain abilities but if all you want are features that are physically built into the OSD it should work.. right?  I ask because I have a drone plastics OSD/PDB for my alien and would love to not have to hook it up to a uart but be able to use the onboard current sensor, rssi and have voltage and a timer?

ShikOfTheRa, 2017-04-14:
Yes, it can do this now... just how you configure and enable/disable features. If you have gps chip you can also hook it up to that.
I run a standalone version in my groundstation for just for voltage display / warning

bfasty, 2017-04-14:
@ShikOfTheRa So do I just not select a FC on the config tab or would it be more complicated than that?  All I want on the screen is volts, amps, Mah drawn, Timer and RSSi.

ShikOfTheRa, 2017-04-14:
Don't select FC for any 
Select the layout closest to what you want then move items around on screen and enable/disable.
If you upload via arduino, in config.h there is a preset called standalone. It's not in the flash from gui, but I'll add that in next few days to make even easier.
It's not to difficult. Don't overthink! What you want is pretty straightforward.

> On 14 Apr 2017, at 14:30, bfasty <notifications@github.com> wrote:
> 
> @ShikOfTheRa So do I just not select a FC on the config tab or would it be more complicated than that? All I want on the screen is volts, amps, Mah drawn, Timer and RSSi.
>

bfasty, 2017-04-14:
I didn't see the standalone feature in the config.h on the current beta release.  I must have missed it.  I look forward to the next beta release with it in there.  
Thanks your getting back to me and I look forward to the release
Thanks again

ShikOfTheRa, 2017-04-18:
Great! new GUI has hex uploader. Standalone isn't one of the configured build options. I'll add it at some point

bfasty, 2017-04-18:
I saw that! the Hex loader worked great!!  Now I'll sit patiently in the corner and wait for standalone mode(twiddling thumbs). 

P.S.  the performance of 1.7.1 is the best I've used so far!!!

Pierre-A, 2017-04-22:
**@bfasty**, for standalone mode, just uncomment the NOCONTROLLER preset (line 54 in config.h rev 1.7).

ShikOfTheRa, 2017-06-13:
@bfasty this is also now available via the chrome GUI - it is listed under default minim boards as Basic OSD - No GPS

closed 2017-06-13 by ShikOfTheRa