ShikOfTheRa/scarab-osd #409: deprecated msp commands

fape, 2017-04-17:
There is some changes in betaflight/cleanflight msp commands. Mostly some commands are deprecated and removed or will be remove soon. See [msp_protocol.h](https://github.com/betaflight/betaflight/blob/master/src/main/msp/msp_protocol.h) and [fc_msp.c](https://github.com/betaflight/betaflight/blob/master/src/main/fc/fc_msp.c)

Used by mwosd:
- MSP_LOOP_TIME: Add some define?
- MSP_SET_LOOP_TIME: Add some define?
- MSP_IDENT: msp_protocol.h: `Consumers of the API (API clients) SHOULD first attempt to get a response from the MSP_API_VERSION command. If no response is obtained then client MAY try the legacy MSP_IDENT command`. but MSP_IDENT was removed from fc_msp.c
- MSP_MISC: add MSP_VOLTAGE_METER_CONFIG. I'm going to send you a PR.

Not used by mwosd:
- MSP_BF_CONFIG 
- MSP_SET_BF_CONFIG
- MSP_BF_BUILD_INFO
- MSP_BOX: defined but not used. Removeable?
- MSP_SET_BOX: defined but not used. Removeable?
- MSP_SET_MISC: defined but not used. Removeable?
- MSP_MOTOR_PINS: defined but not used. Removeable?

And there is a big question:
Mwdosd.MSP_SERVO_CONF and BF/CF.MSP_SERVO_CONFIGURATIONS use the same code: 120.
But mwosd reads/writes only 5 ints: 16,16,16,8,16 (same in baseflight)
but Betaflight/cleanflight uses 6 ints 16,16,16,8,8,32. 
As far I see the MSP_SERVO_CONF is used only for baseflight. So I thin it is a minor issue now.

ShikOfTheRa, 2017-04-18:
Thanks for heads up. Unaware of these changes.

MSP_LOOP_TIME/MSP_SET_LOOP_TIME:
Well, I guess leaving it in will cause no harm if not used by FC. In some cases not visible unless enabled. Do you know what versions will it be removed from in CF/BF? Like to keep in sync if can.

MSP_IDENT:
Its still in the code, however hasn’t been used for some time. Had been thinking of re-introducing it. Been thinking about doing this for some time with MSP_API_VERSION. I have added to the todo list.

MSP_MISC:
Thank you – merged. Question – there is no request sent from MWOSD for 

MSP_VOLTAGE_METER_CONFIG Presumably MSP+MISC will no longer be required and we need to send the REQuest? Do you know what versions will it be removed from in CF/BF?

MSP_BOX/MSP_SET_BOX/MSP_SET_MISC/MSP_MOTOR_PINS:
Will remove as not used

MSP_SERVO_CONF:
Ony use for baseflight fixedwing. I’d like to add for iNav. Will have to consider…

McGiverGim, 2017-04-18:
I think that all this commands were deprecated some time ago, but have been removed in CF2.0 and BF3.1.7. Both have now the same base.

fape, 2017-04-18:
Thanks. Sorry I missed the request for msp_voltagae_meter. I'm going to send a PR soon.
Unfortunately baseflight only use MSP_MISC, But MSP_VOLTAGE_METER_CONFIG is in betaflight long time ago [link](https://github.com/betaflight/betaflight/commit/99089b9e70107269d171f05b3590b1548b65ae8d) same in cleanflight [link](https://github.com/cleanflight/cleanflight/commit/99089b9e70107269d171f05b3590b1548b65ae8d)