ShikOfTheRa/scarab-osd #411: Chrome GUI says "wrong version"

grizewald, 2017-04-17:
So, I wanted to update my flight controller to the latest version. It has an integrated MinimOSD which was running fine on version 1.6 which I'd previously flashed with the Arduino IDE.
I paid for and installed the Chrome GUI but couldn't see the "Flash" button that the documentation refers to. No matter, I still have the Arduino IDE installed. So I configured and compiled version 1.7 and flashed it to the OSD. Now, when I try to connect via the Chrome GUI, most of the time it just connects and then immediately disconnects, or it says that the version of the OSD firmware is incorrect and that I should update it.

This all used to work without problems. Now I can't even configure my OSD!

I'm running the latest Chrome on Ubuntu Mate 16.04.2 (64-bit).

Any help?

ShikOfTheRa, 2017-04-18:
can you confirm teh version of configurator. If can't see in teh GUI unde rteh "mwosd configurator" image then ight click on the chrome app icon to view.

There are a large number of users who seem OK, but have a number of reports raised by ubunto users which we are looking into.

ShikOfTheRa, 2017-04-18:
Ah - just a thought - can you uninstall all - including any free ones etc then re-install the one from chrome store?

grizewald, 2017-04-18:
The GUI version is 1.6.0.9, freshly installed yesterday from the Chrome app store and no other versions of it have ever been installed before.
I also tried flashing 1.7.1 of the OSD code, but no difference.
It would be amazingly helpful if the GUI said "Expected version x, found version y" instead of the utterly opaque "wrong version" message that it currently delviers.

ShikOfTheRa, 2017-04-18:
I honesty have no idea... The version on the store is 1.7.1.1 I have just uninstalled and re-installed to try...
1.6.0.9 hasn't been there for a long time. Have been numerouse updates since the.

Hold fire - will get ubunto installed and test...

Good point on the wrong version message. See if we can improve in that in future...

grizewald, 2017-04-18:
It gets stranger!

I checked on the app store and indeed, it shows 1.7.1.1 as being the latest version of the GUI. So I removed it from Chrome, restarted Chrome and added it again.

Now I have the right version, but it doesn't seem to be working properly. There are no tabs like there were before and I can't select anything in the drop-down that is visible between the buttons labelled "Load online HEX FIRMWARE" and "Flash FW".
Would having a quite small screen (1366x768) upset it? I'll plug into a full sized screen and see if anything changes.

grizewald, 2017-04-18:
In answer to my own question, no. The screen size doesn't make a difference.

This is what I see:

[image]

and if I scroll all the way down, I see this:

[image]

Hope that helps!

error414, 2017-04-19:
Can you try new version?  1.7.1.2 https://chrome.google.com/webstore/detail/mwosd-configurator/jdfbjgomfcfpgkdjfnmekbkhkbgfminf

grizewald, 2017-04-19:
The GUI displays correctly now with 1.7.1.2 installed.

So I selected the "Betaflight 1.7.1" online hex, chose the correct serial port and 115200 baud and said "Flash". The blue LED on my UART-Serial adapter lit up and the app now just sits there saying "Waiting for bootloader reset".

grizewald, 2017-04-19:
It does connect to the firmware that I flashed manually though and I was able to correctly set up my OSD. Nice one!

Do you want me to close this issue and open a new one for the flashing problem, or just carry on here?

grizewald, 2017-04-21:
I'll close this issue as I can now connect to the OSD.

Thanks for the help!

closed 2017-04-21 by grizewald