ShikOfTheRa/scarab-osd #413: Inverted AHI pitch

fantasiiio, 2017-04-18:
Hi,

I compiled MWOsd for a good old APM 2.6 and minimosd. All seem to work good exept for the pitch of AHI is inversed. When the nose goes up, the horizontal bar goes up. Same thing for nose down. 
If I try it in mission planner for apm all is correct. My apm camera gimbal also is correct. I tried the compiler flag INVERT_PITCH, but it does nothing.

So, do you have a hint of why it does that ?

ShikOfTheRa, 2017-04-18:
That is normal operation for FPV fixed camera - the ahi should track horizon. Viewing with a camera on a gimbal, I can understand why might prefer the other way. I can't verify INVERT_PITCH today - will try tomorrow

fantasiiio, 2017-04-18:
Yes, the AHI is supposed to track horizon, but it's not. When the nose goes up, horizon goes down, but AHI goes up.. it's inversed.

ShikOfTheRa, 2017-04-18:
Ah sorry - yes - I misread on my phone...
I will check for you tonight. 

I have never used APM - it is quite possible it is wrong, but no-one has raised to me before. Maybe they don't use ahi!

fantasiiio, 2017-04-18:
I just want to specify that I tried the gimbal only to see if it works correctly. I will not use gimbal and ahi at same time because the horizon doesnt move and the ahi is not useful.

If you want I can try to debug the code. I'm an experienced programmer.

fantasiiio, 2017-04-19:
I made a small pull request

closed 2017-04-24 by fantasiiio

Kinder0083, 2017-06-28:
Hello

I have the same problem with my Pixhawk.
The pitch is inverted.

Do you have a solution ?

Thanks

Grèg

fantasiiio, 2017-06-29:
I have a solution that work very well.

modify file screen.ino
in function displayHorizon (line 443 in my version),
replace
```
#ifdef INVERT_PITCH
  if(pitchAngle>0) 
#else
  if(pitchAngle<0)
#endif	  
  {
    screenBuffer[1]='-';
    offset++;
  }
	  
```
by
```
#ifdef INVERT_PITCH
  pitchAngle = -pitchAngle;
#endif	  
  if(pitchAngle<0)
  {
    screenBuffer[1]='-';
    offset++;
  }
```
then in config.h uncomment the line
`#define INVERT_PITCH`

That's it

Kinder0083, 2017-06-29:
Hello

I will testing that this afternoon.

Thanks

ShikOfTheRa, 2017-06-29:
Hello - I just submitted a couple of changes this morning which may help.
If I get more feedback about pixhawk being wrong direction I'll change in code. I'm surprised not raised before

//#define REVERSE_AHI_PITCH         // Reverse pitch / roll direction of AHI - for DJI / Eastern bloc OSD users
//#define REVERSE_AHI_ROLL          // Reverse pitch / roll direction of AHI - for DJI / Eastern bloc OSD users
//#define INVERT_PITCH_SIGN         // Invert the sign of the displayed numeric value for the pitch angle (ex: pitch up = positive )
//#define INVERT_ROLL_SIGN          // Invert the sign of the displayed numeric value for the roll angle (ex: roll right = negative )

Kinder0083, 2017-06-30:
Hello

Your solution works verry fine.
If you have any question about pixhawk with osd or other equipment, ask me.

Thanks

Grég

ShikOfTheRa, 2017-06-30:
Thanks for feedback. I might have a few questions. 

It sounds like we need to change this. Ideally I'd like one more to report / confirm. I would have thought a few more would have raised something so visible.

Just to clarify, "correct" operation should be plane nose down AHI move up on screen / goggles.
I ask because for some they want opposite.. Mainly CRT fliers who want to know attitude of plane rather than where horizon is (I think!)

Kinder0083, 2017-06-30:
Hello

I'm a flight engineer on French plane and for me the good representation is
the center cross which represents the nose of the plane and the 6 points (
3 right of the center cross and 3 left of the center cross ). 6 points
represent the artificial horizon.

After changed your code in "screen.ino" the representation is good. When my
plane increase altitude with the nose up, the cross is upper the horizon;
inverted with descend of my plane.
When I turn left the horizon tilt right, inverted with turn right.
I joint a photo wich represents a right turn.

For a good FPV you must to imagine your position in the cockpit.

Say me if I not answer at your questions.