ShikOfTheRa/scarab-osd #416: Add a CLEANFLIGHT2 define to the Config.h file

McGiverGim, 2017-04-27:
The Cleanflight 2.x code now is exactly the same as Betaflight 3.1.7, so it must act exactly as the BETAFLIGHT (or BETAFLIGHT31) definition.

The easier solution is to add a `#define CLEANFLIGHT2` to the `Config.h`, and maybe a:
```
#ifdef CLEANFLIGHT2   
  #define BETAFLIGHT31
#endif
```
to the `Def.h`.

But I don't know if this is a correct solution, o is maybe cleaner modify all the conditions where the BETAFLIGHT31 appear to add the CLEANFLIGHT2 too.

McGiverGim, 2017-04-27:
I've seen too that now in `Def.h`:
```
#ifdef CLEANFLIGHT    //set up latest at time of release
  #define CLEANFLIGHT190
#endif
```
So I suppose that `CLEANFLIGHT must change to point to the new CLEANFLIGHT2 definition too.

ShikOfTheRa, 2017-04-27:
Yep - planning to do a bunch of changes like this next week.

ShikOfTheRa, 2017-06-13:
Bit longer than a week - but its done in main repository. GUI hex distributions will be updated after made a few more additions. Enough to release..

closed 2017-06-13 by ShikOfTheRa

McGiverGim, 2017-06-13:
Thanks! I will test it ASAP ;)

ShikOfTheRa, 2017-06-13:
So normal CLEANFLIGHT define invokes v2 support. If want earlier ones need to enable CLEANFLIGHT190