ShikOfTheRa/scarab-osd #418: Do not load fonts version 1.7.1

kak-dela007, 2017-04-30:
The firmware downloaded through a paid GUI (version for APM) , but to upload a font does not work. On the screen appears something like this. How to fix it? 
[url=http://radikal.ru][img]http://s009.radikal.ru/i307/1704/34/de99064a7056.jpg[/img][/url]

moggiex, 2017-05-03:
This is exactly the same issue I am having. See this video 

[https://www.youtube.com/watch?v=2zYQsMgFMEU](https://www.youtube.com/watch?v=2zYQsMgFMEU)

Matt

Creautre, 2017-05-04:
I tried using EEPROM Clear on Arduino IDE and reflashing the firmware
 over and over again but it never worked [ desoldered from my Circuit of
 course ].
I managed to fix the Issue by Using the "FLASH" function on the top right of MWOSD Configurator instead of using "Connect". I first uploaded ClearEEPROM, then Bold font installer and then "Default minim Betaflight 1.7.1".

Check my Thread about this

http://intofpv.com/t-microminimosd-not-programmable?page=2

moggiex, 2017-05-04:
Yep that approach worked, thanks for sharing. I appreciate it!

ShikOfTheRa, 2017-05-10:
Hi all - sorry for slow response - tied up at work at moment- 
moggiesx - have watched many of your vids. Think last one was your finwing traveller which I know have waiting for chance to do something.

Re the fonts - are you guys using FC passthrough for programming?

Creautre, 2017-05-10:
@ShikOfTheRa As you can read in the Thread I posted I first had the issue when I used FC passthrough. After that I managed to fix everything when I had the MicroMinimOSD desoldered and directly connected it for programming and used the "FLASH" function. The OSD worked perfectly fine afterwards but I decided to put it to the test and (while still desoldered from the copter and was still directly connected) I used the font upload feature and it immediatly had the same issue again. 

So the process was 100% repeatable. Only fix for me was to use the Flash function of the OSD Configurator. The normal Font Upload function while you are connected always messed up the Fonts

moggiex, 2017-05-10:
@ShikOfTheRa, no worries, work pays for models :)

Clearing the eeprom, flashing the fonts, then flashing the firmware appears to work well and as @Creautre mentions above, 100% repeatable error using the GUI for loading the fonts.

Working now with the process above. Well kind of, when flashing with Pixhawk set as the FC the flight mode indicator is not changing, but should be in a different thread.

PS. The traveller plane. Mix in either some rudder with the ailerons and/or add washout on the ailerons too. Also differential thrust would also help with that models adverse yaw issues. I swapped mine for a PNF Vampire, so super happy now :D 

Matt

ShikOfTheRa, 2017-05-10:
OK thanks both - will pass on to dev. Must admit I haven't seen this issue, but can understand how it might occur.

That Traveller could be awesome. I hope they do a V2...

steph280, 2017-05-10:
Same problem here, and the steps from Creautre did resolve it.  Thanks!

ShikOfTheRa, 2017-05-11:
Thank you for reporting. Confirm bug found. Updated configurator being deployed to store today.

Estimate this will be available 12:00 UTC 11/05/2017 
Configurator Version 1.7.1.5

ShikOfTheRa, 2017-05-15:
This has been resolved and confirmed by one of users

closed 2017-05-15 by ShikOfTheRa