ShikOfTheRa/scarab-osd #419: Problems connecting to the APM

kak-dela007, 2017-05-04:
1. Not correctly displays the gas level. Responds to stick "ruder" instead of reagirovat on "throttle" . When you download the firmware through Arduino IDE I tried to activate the following lines in different combinations, but it does not give results.
#define AUTOTHROTTLE 
//#define HIGHTHROTTLE 1900                // Maximum recognised value for throttle 
//#define LOWTHROTTLE  1100                // Minimum recognised value for throttle

 2. Not working correctly the horizon in pitch. Row changes do not work.

#define INVERT_PITCH              // Invert the sign of the displayed numeric value for the pitch angle (ex: pitch up = positive )
#define REVERSEAHI                // Reverse pitch / roll direction of AHI - for DJI / Eastern bloc OSD users

 3. How to display vertical speed? 
4. How to display distance to home?

ShikOfTheRa, 2017-05-10:
Are you using 1.7?
1:
 Channel order matters - it is configured for default of AERT
If you are not using this, need to use arduino and upload with correct tx enabled in config.h.. Longer term will look to make that GUI configurable. Memory permitting

//#define TX_RHF                    // Enable for Robe/Hitec/Futaba (Only use for GPSOSD/NAZA/APM/PIXHAWK/MAVLINK)
//#define TX_GS                     // Enable for Graupner/Spektrum (Only use for GPSOSD/NAZA/APM/PIXHAWK/MAVLINK)    
//#define TX_M                      // Enable for Multiplex (Only use for GPSOSD/NAZA/APM/PIXHAWK/MAVLINK)
//#define TX_HS                     // Enable for Hitec/Sanwa (Only use for GPSOSD/NAZA/APM/PIXHAWK/MAVLINK)

2:

ShikOfTheRa, 2017-05-10:
3: No option to display climb rate. There is a climb rate indicator. I can look at add if this is a request?
[image]

ShikOfTheRa, 2017-05-10:
4:
[image]

ShikOfTheRa, 2017-06-13:
Default channel order changed for APM users. Can be overridden if required.
Thsi is available in main repository if using Arduino to upload. Will also be available in next pre-complied hex for GUI

closed 2017-06-13 by ShikOfTheRa