ShikOfTheRa/scarab-osd #420: Failsafe indicator on screen

SergeantSeven, 2017-05-07:
Hello, 
This is a feature request.
I'm interested in having the ability to see an indicator that the craft is in fail safe.
It would basically be the same as the arm/disarm status but would actually read "FAILSAFE" as a status.
Reading through the available MSP commands I'm not sure how this would be implemented but I feel its would be valuable to many pilots to know when the flight controller has started Failsafe.

ShikOfTheRa, 2017-05-10:
There is an option in there now. CAn edit layout to show APSTATUS as per image:

[image]

SergeantSeven, 2017-05-14:
will this work with cleanflight without an auto pilot mode and just a failsafe condition?

ShikOfTheRa, 2017-05-14:
Not 100% sure. Best way is to try to simulate and find out.

SergeantSeven, 2017-05-15:
Doesn't seem to work that way for me. I'm assuming AP status only works with flight controllers with Auto pilot functions, since cleanflight does not support this functionality in that degree, I am also assuming "FAILSAFE" is not included in AP status commands sent by any flight controller. please correct me if im assuming wrong. I did activate "Gps status" and located the AP status on the layout where I'd want it. but cleanflight does not seem to send anything out for this to be displayed. My RSSI indicator does flash and show 0% when in failsafe. I guess this would be acceptable for the time being, but an exclusive failsafe indicator would be beneficial still, especially for those who do not have RSSI available to them.

ShikOfTheRa, 2017-05-17:
Looked at code. confirm not supported. I will add shortly.

in serial.ino, if you find this line:
      case 10:
        mode.gpshome |= bit;

Change to:
      case 10:
        mode.gpshome |= bit;
      case 27:
        mode.gpshome |= bit;

temporary hack. It will display AUTO RTH, but you can change that as a temp fix also:
in globalvariables.h:
const char APRTHtext[]      PROGMEM = "AUTO RTH";

SergeantSeven, 2017-05-18:
Sounds good, I'll give the hacks a try, thanks!

ShikOfTheRa, 2017-06-13:
This has now been added to main repository for upload by arduino. Will be in next GUI compiled hex release

closed 2017-06-13 by ShikOfTheRa