ShikOfTheRa/scarab-osd #422: Platformio env names vs Platformio ide

fape, 2017-05-11:
Platformio ide does not find any environment. I got "empty projects" warning.
Because env names contain `'.'` (dot) character. [Only `[a-z0-9_]+` is allowed in env names.](http://docs.platformio.org/en/stable/projectconf.html#projectconf-section-env). If I replace all the dots in `platformio.ini` to `'_'` and everything works fine in the ide. Unfortunately it works from command line with dots :S

I can create a PR with new/replaced platformio.ini. But I think it may need lot more changes eg from configurator side as well. (Or replace `'_'` back to `'.'` in `.travis.yml` ?)

ShikOfTheRa, 2017-05-11:
Actually I only use it to build hex files - so cli ok for me! 

At some point I will use because not all environments compile - and want to make it compatible with both. If not in a hurry I will make all changes in about 3 weeks?

fape, 2017-05-11:
Thanks. I can live with my workaround some more weeks ;)

ShikOfTheRa, 2017-10-31:
meh, sorry about this, but the re-work of the GUI to accommodate means not going to be able to implement.

It was an unknown consequence of the decision to use "," as separator, although never thought anyone would use the files for normal IDE use. Was only ever intended to create the hex files for deployment files.

fape, 2017-10-31:
No problem, take your time. I can live with the workaround (replace all) but basically now I'm busy with other drone related development: eg  [voroshkov/Chorus-RF-Laptimer](https://github.com/voroshkov/Chorus-RF-Laptimer)

ShikOfTheRa, 2017-10-31:
Ok I'll close this as trying to close out all issues ready to restart developement- but have put it on the "To do" list managed on "the wiki

ShikOfTheRa, 2017-10-31:
Very cool project. I will look now. I always wondered if possible to switch frequencies and monitor fast enough - or if have to have separate rx for each frequency.

closed 2017-10-31 by ShikOfTheRa