ShikOfTheRa/scarab-osd #424: Current reading is multiplied by 10

skaman82, 2017-05-17:
I get mAh readings multiplied by 10 in the osd compared with FC readings: 
FC is showing 3 mAh consumed in Configurator and MWOSD is showing 30 mAh. 

Is there a fix for that?

damonp, 2017-05-17:
I ran another pack through last night using PR#421 and the battery charged back to within 5% of what the OSD displayed as total MaH.  The `current scale` in Betaflight I ended up with was `770`.  Not that that should match anyone else's as I'm sure all current sensors are different.  I was just surprised it was nearly double the default.  

@skaman82 What are you using for `current scale` in Betaflight?

skaman82, 2017-05-17:
@damonp my current scale is 271 and the Configurator and I can see the correct mAh value in the Configurator App - but in the OSD it is still multiplied by 10. Running on latest public release of Release 1.6

ShikOfTheRa, 2017-05-17:
Which FC - betaflight?
Also how is it connected - to FC or direct to OSD.

Sure there was a bug fixed after 1.6 somewhere for this issue. Might be a small tweak to save updating to 1.7 if not needed

skaman82, 2017-05-17:
@ShikOfTheRa Using INAV 1.7 on SPRacingF3 EVO FC. Current sensor is connected to the FC and calibrated using the current readings in the INAV configurator.

I also have purchased the 1.7 MWOS GUI - but I disliked the GUI so much and so I am using MWOS 1.6 again :)

skaman82, 2017-05-17:
@ShikOfTheRa BTW. tested this today in MWOSD 1.7 with a beerotor F3 FC and PDB. It also doesn't work correctly. I'm getting 38 mAh in configurator but in the OSD I see 0 mAh. Different outcome but still not working like it should

McGiverGim, 2017-05-18:
I don't know if this has something to do, but since BF 3.1.7/CF2.0 the current reading is now in centiamperes, before it was in deciamperes. At least in the blackbox log.

digitalentity, 2017-05-18:
Hm... I probably need to set up something with external OSD hardware and test myself.

skaman82, 2017-05-18:
[image]

skaman82, 2017-05-18:
My settings
[image]
[image]

skaman82, 2017-05-18:
Update: 1.7.1 ist showing correct results

damonp, 2017-05-18:
Nice. Did you need to set the mwosd current scale to 1?  Wondering if my setup is the odd one.

closed 2017-06-13 by ShikOfTheRa