ShikOfTheRa/scarab-osd #425: MWOSD 1.7 GUI  without FTDI? (OSD already installed...)

skorzo, 2017-05-20:
Hi!

I flashed Mwosd 1.7 and configured it with MWOSD Configurator GUI.

Probably I forgot to install the fonts and now I have lots of bad characters on my display, but the FC readings are OK.

Since I have the osd already soldered and hidden inside my quad frame,  Is there any way to re-flash the fonts or connect to Configurator without desoldering everything and setting up the FTDI?

I mean something like a passthrough with the flight controller so I can access to the OSD.

Thanks!

error414, 2017-05-24:
Hi,

you can check the video https://www.youtube.com/watch?v=Q6CXKAYYSB4 . 

I tested serial passthrough  with baudrate 115200 and dedicated UART, I have no good experience with soft serial, it is quite unstable. 

And it is important to have same baudrate for PC <-> FC and FC <-> MWOSD.

skorzo, 2017-05-26:
Solved!! Many thanks for the tip, that works.

closed 2017-05-26 by skorzo