ShikOfTheRa/scarab-osd #426: GUI chrome store price is not available

amin005, 2017-05-21:
price is not available so no GUI for ver 1.7
Ithink you are killing this project

MathijsG, 2017-06-07:
What if you use the chrome extension in for example this latest test version: https://github.com/ShikOfTheRa/scarab-osd/archive/1.7.1.1.zip

it's within the MW_OSD_GUI folder, and you can manually drag it in Chrome on the extension page of Chrome.

That's how I bypassed the paid option. Not because I don't want to pay for this beautiful project, but because Google doesn't let me. Outside the US credit cards are much less common.

ShikOfTheRa, 2017-06-13:
For Price not available see below.

Paypal option is actively being worked on at moment.

30 Unable to load MWOSD GUI configurator from the google chrome store

Unfortunately Google has not made its services available in your country. The following approach can be used to make it appear that you are in another country.
Download TunnelBear from the Google Chrome Store.
Launch the app and create an account.
You'll see the switch to turn TunnelBear on and a map of the world. Simply tap a tunnel anywhere on the globe (suggest US) and wait for the bear to tunnel its way there.
If you get a connection request, agree to it.
Once TunnelBear has created a connection to your new location, you can install MWOSD. Note : If the store still says it's not available in your country, you may have to clear clear the cache for the Chrome Note : Upgrades may require you to re-run TunnelBear
31 Can I use Paypal to pay for configurator

At some point in the future we hope to have an option for direct support of PayPal
For now, this is a decision by Google which does allow in some countries.

ShikOfTheRa, 2017-08-02:
For windows users, it is available here. MAC will come in future once PC testing complete
https://www.virtualpilot.co.uk/index.php?route=product/product&path=62&product_id=53

closed 2017-08-02 by ShikOfTheRa