ShikOfTheRa/scarab-osd #427: Unable to configure

OtterNas3, 2017-05-25:
As me, and i guess many others, cant figure out why Google is not accepting the Credit Card in the Chrome App-Store, is there a reason why PayPal is not accepted?

OtterNas3, 2017-05-25:
I would donate even 5$ for the Configurator if i would be able to get it another way...

ShikOfTheRa, 2017-06-01:
I don't know why google doesn't accept . It is a pain.
A version that can accept paypal is coming soon hope within a month

MathijsG, 2017-06-07:
What if you use the chrome extension in for example this latest test version: https://github.com/ShikOfTheRa/scarab-osd/archive/1.7.1.1.zip

it's within the MW_OSD_GUI folder, and you can manually drag it in Chrome on the extension page of Chrome.

That's how I bypassed the paid option. Not because I don't want to pay for this beautiful project, but because Google doesn't let me. Outside the US credit cards are much less common.

OtterNas3, 2017-06-07:
Unfortunately u can't use that GUI for the v1.7+ of the OSD Firmware

OtterNas3, 2017-06-23:
Hey Shik,

sorry for bothering you again, but, any Update on this?

-Otter-

ShikOfTheRa, 2017-06-26:
Its pretty close. Drop email to gary@flypix.co.uk as will need a test group soon

closed 2017-08-07 by ShikOfTheRa