ShikOfTheRa/scarab-osd #428: Hardware requirement

hexagon-robotics, 2017-06-04:
Hello!

I am designing a flight controller for racing. I have already implemented a Video / Audio Transmitter. But I would like to include an OSD.

What hardware do I need to integrate scarab-OSD on my board?

Best regards,
Javier

MathijsG, 2017-06-07:
If you look at the many flashing tutorials of a MinimOSD chip, you discover that it often is a ATMega328 5V 16Mhz.

hexagon-robotics, 2017-06-07:
Thanks Math!

I have seen that ATMega328 (AVR) is the classic to OSD for FPV. I have also seen other OSD projects with 8bits uC's as PIC or ARM.

But MAX7456 is obsolete. I have asked to Maxim a few weeks ago. 

scarab-OSD support other chip?.

I have seen a alternative, the AT7456 , I think that it's made by a company in china but I not sure. Does anyone know him?.

Sorry my english.

Regards.

error414, 2017-06-08:
Hi Javier,

I dont know AT7456, however developer of scarab-OSD created own OSD HW, and he used AB7456, see http://www.virtualpilot.co.uk/index.php?route=product/product&product_id=51

`Uses the newest AB7456 chip – generates far less heat and uses far less power than MAX7456 based clones`

hexagon-robotics, 2017-06-11:
Hi error414,

it sounds interesting, thanks very much!

It's the piece I'm missing to finish the flight controller

You know where I could get some pieces. Even for prototypes.

Best regards,
Javier

hexagon-robotics, 2017-06-11:
Sorry,

The chip has the Airbot logo, then The [chip](https://cdn6.bigcommerce.com/s-xkoep7/images/stencil/1280x1280/products/903/6357/Micro-OSD-v23-b__72686.1484124255.JPG?c=2) was created by Airbot?

ShikOfTheRa, 2017-06-13:
I think it is same as AT7456 but created for Airbot. Airbot can manufacture boards for you too

hexagon-robotics, 2017-06-13:
Yes, is custom model by Airbot, yesterday I contacted with Airbot.

I use the same architecture, ARM Cortex M4F, but manufacturer of uC is different, is Motorola Freescale - NXP - Qualcomm.

180Mhz, 2Mb Flash, 512Kb RAM in BGA format 9x9 mm

I know a manufacturer en Shenzhen who contacted with me time ago, but I have nothin closed. I look for good price and good quality on the finish.

Regards

ShikOfTheRa, 2017-06-13:
yes speak to Li at airbot. Maybe he can help.

Also many sources like following:
http://www.icofchina.com/pro/kaiguan/10.html

hexagon-robotics, 2017-06-13:
Thanks a lot ShikOfTheRa!

U are big.

closed 2017-06-13 by hexagon-robotics