ShikOfTheRa/scarab-osd #429: APM - Distance To Home incorrect

gsimpson2g, 2017-06-07:
I upgraded to MWOSD 1.7 today, from the original MinimOSD firmware. I noticed my Distance To Home value is inaccurate. One test showed me 2200ft from home when I was actually about 4800ft from home. The direction to home arrow also appeared to be inaccurate.

I will do some more testing to verify.

ShikOfTheRa, 2017-06-12:
Can one of you try new hex (Listed under TEST section in GUI) or code direct from repository.
Currently resets altitude and home first time it is armed with enough sats.

gsimpson2g, 2017-06-13:
I will give it a go tomorrow.

gsimpson2g, 2017-06-16:
Tested it today. It looks to be showing dist from home around half what it really is. OSD was showing me a distance of 3300ish ft, yet I know I was closer to 6600ft. The total distance traveled looks to be accurate on summary screen (showed 12800ish ft traveled).

Here is the video of the 6600ish ft showing incorrectly as well as the arrow to home showing wrong. https://www.youtube.com/watch?v=XazcgH1YaBY

ShikOfTheRa, 2017-06-16:
Thanks for posting the video. I love looking at them and it helps me identify issues. Also I think not many use for APM so extra bonus to see that.

I had a look in code and I see for some reason in APM, the Longitude scaling is disabled. Not sure why - I think this must have been by mistake. At equator - 1 degree longitude = 111km. at 45 degree it is 75km

I wonder if this is the cause. I have re-enabled in the code. Its in repository - or can use this hex in GUI "Test MultiRotor APM PIXHAWK"

By any chance would you have a tlog for the flight you could share?

ShikOfTheRa, 2017-07-20:
Were you able to test?

dawiemalan, 2017-07-24:
Had a similar issue with a flight yesterday. The home position changed during the flight. There was no disarm event in the logs and no obvious GPS glitches. My GCS froze when saving the tlog so I don't have that, but I can provide a PX4 bin log, and a short video clip of the time when this happened.

ShikOfTheRa, 2017-07-24:
vide / data would help. 
For distance issue:  was that the latest version installed from GUI configurator?
For GPS home reset:
1, Restart of OSD. should be obvious on flight video
2, 3 consecutive packets of mavlink data saying disarmed. 

//#define RESETHOMEARMED            // Uncomment this ONLY if armed information is sent within telemetry feeed AND you do not want to reset home position when re-arming. DO NOT DISARM IN FLIGHT

dawiemalan, 2017-07-24:
[log_5_2017-7-23-16-19-22-fx61.zip](https://github.com/ShikOfTheRa/scarab-osd/files/1169242/log_5_2017-7-23-16-19-22-fx61.zip)

Yes, firmware is Fixedwing APM PIXHAWK 1.7.1 installed with Chrome GUI.
Here is the video:  https://youtu.be/3AXS4X7-3T4
The OSD did reset just before launch, and upon landing (which was a bit rough). I've got the VTx, OSD, FPV and HD camera as well as 500mW telemetry radio on a separate 5A BEC, with an LC filter after the BEC.

ShikOfTheRa, 2017-07-24:
Can you tell from vid when it happened. Not knowing you're location etc., as I watch the vid I can't see where it is wrong

dawiemalan, 2017-07-24:
At 4:58 you can see it shows the home position 314m directly to the right. This is more or less correct. 1 second later it show home direction more forward, and distance starts to reduce, even though direction of flight it still the same.  At 5:24 it shows home distance as 2m, while it is 450m or more away. Home direction should be SE.  This completely disoriented me, which is why I did a circle there.
At 5:39 the nose is pointing directly at the launch point (the dot towards the lower left of that open field is my car). Home arrow shows way off toward the right, distance 198m.  Should have pointed straight ahead.
At 6:11 it again shows it is at home position (3m) but now it is at yet another position, guessing about 400m away from previous (incorrect) home position.
7:25 another new home position.
At 8:44 it almost on top of the real home position, but shows 60m towards the right (we're now about 900m away from the positions above).
9:30 is more or less correct, you can see my car almost straight ahead. Distance also about right at a guess.

ShikOfTheRa, 2017-07-24:
Thanks for that. Its hard to visualise not knowing the area. It does look a little odd in places. I am going to have to build a pixhawk / apm unit to test for myself. Its unfair to keep asking you guys.

The latest version (released this morning) 1.7.2 contains the bugfix associated with a distance issue. I hadn't considered it but It will probably affect direction to a certain degree too.
Suggest to load that version and retest. But keep aware of home point! 

I fly quite a lot with OSD for testing and home arrow / distance has been good for years, but so far not personally testedwith APM/PIXHAWK setups. 
 
From personal experience the following can make compass 180 degrees out..
1 if using GPS for direction Wind - can make the home arrow very weird. As I see the plane turn can see quite a variation in airspeed
2 if using magnetic heading even if calibrated interference from current in cables can have a huge impact. This one still amazes me sometimes how much..
3 Direction can lag a little. It depends upon update speed of FC to OSD. I do notice it on a turn.

dawiemalan, 2017-07-24:
Thanks for your quick response and support. Some comments that might help:
1. Yes, it was a bit windy, around 6m/s.
2. The compass display looks fine throughout the flight, but seems to be GPS heading, not magnetic. At takeoff the heading is wrong, but quickly corrects once it has some speed. I was considering disabling magnetic compass, will that have an effect on the OSD? Arduplane is not dependent on magnetic heading AFAIK.
3. Could this issue be related to distance from home? It seems to be fairly accurate closer to the home position.

ShikOfTheRa, 2017-07-24:
1 I could see! I lost a plane 8 years ago because of lack of experience. When heading home windspeed being greater than airspeed = home arrow reverse so you turn around and go wrong way and then home arrow works and tells you to turn around. So end up in circles going ever further away. Ni got plane back a week later so all good!
2 If gets heading from data send in VFR_HUD packet - I believe this is supposed to be compass but not 100% sure. Near home, heading home arrow is poor. Ignore anything less than 5-10m  
The heading displayed should be magnetic heading not GPS. The GPS heading for mavlink/px4 is currently ignored.
3 If in an EW direction. It is actually scaling of longitude. For this original issue raised it worked out about right. It was in the code but I accidentally removed it during some testing.

Try and see if it solves the issue. I will progress and build a copter for testing..

UebertreibeR, 2017-08-13:
Hi,
I'm having the same Issue with my APM and MWOSD 1.7 installed about 5 weeks ago. The GPS Home distance just isn't showing the right numbers.

Unfortunately I don't see a "Test" section in the GUI to try your fix. Also if I flash a new Update with the GUI - what happens to the changes I've made in the config.h and Def.h   in my previous firmware?I only updated using arduino app so far.

ShikOfTheRa, 2017-08-14:
As long as not using PX4, install 1.7.2.0 form here:
https://github.com/ShikOfTheRa/scarab-osd/releases/tag/1.7.2.0

Source code zip for this version is included if you want to configure your own variant, otehrwise can just use latest gui configurator.
Any of your own changes to config.h/def.h will be overwritten by whateve ris in teh new version

ShikOfTheRa, 2017-10-31:
Late reply... thanks for feedback from testing

closed 2017-10-31 by ShikOfTheRa