ShikOfTheRa/scarab-osd #432: Acro/level mode distorts OSD

Markuzzzz, 2017-06-14:
Hey Guys, I am testing the MW Osd for Kiss. I have a strange error going on. When ever I use level/acro mode selection using a switch on my TX the osd disappears entirely. Not matter on which Aux I set level/acro mode in the Kiss Gui. Anybody having the same problem? Shouldnt it react only on AUX4?

Tried 1.6.8 and the newest beta of 1.7 for Kiss from github.

dustin, 2017-06-14:
This will happen for a couple reasons:

1. You are browning out your max7456 chip (dipping below 5V will reset its state).
2. You've got an OSD switch configured that happens to match what's being sent.

The second condition could also be met by an MSP bug coming from KISS.

Markuzzzz, 2017-06-14:
Pretty sure the first reason isn't the case. OSD doesn't restart and doesn't go to start menu/screen, but goes back to the normal osd screen. I figure it's the latter, in that case there ain't nothing to do but wait for a fix by the KISS team?

ShikOfTheRa, 2017-06-15:
For KISS, IIRC MWOSD only supports Screen switching from an RC channel, 

So if only want same layout all time, can set it in GUI to "Using FC OSD_SW"

Otherwise, maybe try changing the OSD ch value just to see.

[image]

Markuzzzz, 2017-06-15:
Tried using different channels but still the same.. thing is I don't want screen switching but it always does on acro/level switch on aux port.

ShikOfTheRa, 2017-06-26:
Had a look at code this morning and kiss does send mode control. There are two layouts associated with it.

So in brief, easiest way is to set all screen layouts to be the same. 

Change all layouts to be same as per pic (if three show, change all three)
[image]

closed 2017-07-04 by ShikOfTheRa