ShikOfTheRa/scarab-osd #433: Custom placement of a persistent character

4cellOstrich, 2017-06-15:
Hello,

I was wondering what the procedure is for displaying any char from the max7456's set on the osd. I really just want to hard-code in an immobile horizontal bar on either side of the screen!

Thank you,

Owen

ShikOfTheRa, 2017-06-16:
Try this for starters:

Add this in after  displayTime();
displayBars();

Add this into screen.ino :

void displayBars(){
  screenBuffer[0]=0x2d; // where 0x2D is character font "-"
  screenBuffer[1]=0;
  MAX7456_WriteString(screenBuffer,(7*30)+1); // where 7 = row 7 and 1 = column 1
  screenBuffer[0]=0x2d; // where 0x2D is character font "-"
  screenBuffer[1]=0;
  MAX7456_WriteString(screenBuffer,(7*30)+28); // where 7 = row 7 and 8 = character 28
}

4cellOstrich, 2017-06-16:
so I tried that, but I only got a completely blank screen, that said,  I am using a pixhawk on px4, which I know isn't compatible, but I was just using the minimosd board to display these chars, which in theory arent dependent on the pixhawk right?

ShikOfTheRa, 2017-06-16:
Oops try 7*30 instead of 730

730 will be of screen! And pids lock it up

4cellOstrich, 2017-06-16:
It worked! Clutch man!

ShikOfTheRa, 2017-06-16:
That's great.now just play around with positions. I just had a rough stab to get you started

closed 2017-06-21 by ShikOfTheRa