ShikOfTheRa/scarab-osd #434: Audio variometer support

Schalonsus, 2017-06-16:
Would it be possible to use MWOSD and OSD hardware to create an audible variometer which connects to the audio "IN" from the video trasmitter?

ShikOfTheRa, 2017-06-16:
I wondered when someone would ask!

Yes I think so. I think Arduino function uses a lot of memory so might have to look at 
I was thinking about using it as a modem  for antenna tracker, but never had enough time to try it.

I could put something together for you to try - maybe use current sensor or RSSI pin - or would you prefer another?

Schalonsus, 2017-06-17:
Nice! 
Preferably i would like to use the RSSI pin

MarenB, 2017-06-17:
Great, I would sign in as second tester!

ShikOfTheRa, 2017-06-17:
Get your hardware ready then guys. I'll try to get completed in the week.

MarenB, 2017-06-17:
Alright, I'll hurry up finishing the plane.

Maybe this will get myself a little more time: I was wondering if it makes sense to combine the audio output to specific OSD screens? That way it would be possible to activate the audio feedback only when looking for thermals and one wouldn't be annoyed by the useless beeping while ascending with motor power?!

Schalonsus, 2017-06-17:
Or sound is only available while throttle is zero

MarenB, 2017-06-17:
> Or sound is only available while throttle is zero

Also a good idea! Additional to my idea, the sound could be defined just like an OSD paramater similar to voltage, etc. - so anyone can activate or deactivate it in the screens he wants. A coupling to zero throttle would still make sense, although this still wouldn't cover flight modes in which the FC controls throttle autonomously (althold or FBWB, i.e.).

ShikOfTheRa, 2017-06-21:
Had a stab at this. ready for test / debug. I haven't looked at hardware needed yet but guess that's on web somewhere and needs a resistor / capacitor.

/********************       Vario / climbrate Settings         ************************/
#define DISPLAYVARIO                // Enable this to display Vario slider (if climb rate selected in layouts)
//#define DISPLAYCLIMBRATE          // Enable this to display climb rate (if climb rate selected in layouts)
//#define VARIOALARM 150            // Enable this to for flashing climb rate warning. Value in cm/sec (if climb rate selected in layouts)
#define AUDIOVARIO A3               // Enable this for audio vario on RSSI pin A3
//#define AUDIOVARIORC 1200         // Enable this to disable audio vario for throttle values above x uS

Schalonsus, 2017-06-22:
Barometer needed or is GPS sufficient for the code?

ShikOfTheRa, 2017-06-22:
//#define DISPLAYCLIMBRATE
//#define VARIOALARM 150 

Above new additions are tested

For audio...
GPS should be fine if FIXEDWING specified..

This is first cut - suggest if build some hardware can test using GUI simulator. I honestly have no idea if it works yet. Chances are it will need some tweaking, but its ready to test!

Followi9ng as starter for a circit. I havent tested yet..
[image]

Schalonsus, 2017-06-22:
First test done:
I soldered a voltage divider across RSSI (22kOhm) and GND (2,2kOhm). This should result in a 0,5V signal. This is fed to the "audio in" of my transmitter.

But i just get a solid tone if rising altitude in configurator. It starts 3 seconds after start rising and ends 10 senconds after stop rising. Descend makes no tone.

My KK Vario dont need the capacitor. Its Pin2 or Pin3 on the Arduino Nano, so must be digital pin.
Is the A3 pin of the OSD analog or digital?

ShikOfTheRa, 2017-06-22:
Ah - I just uploaded missed commit.
Its positive to get a tone. That's the bit I was unsure of actually. The rest I am confident we can fix.

Give ita try now. A soon as you said nothing on descent I knew I hadn't posted the last...

ShikOfTheRa, 2017-06-22:
And interesting on the capacitor. 
That probably depends what is on input of VTX. If it already has one or not. Not really familiar with how these things are built.

Also, once we have basics of this running we can tweak following settings:
  #define VARIOFREQCLIMB 800      // Climb tone Hz
  #define VARIOFREQSINK 600       // Sink tone Hz
  #define VARIOTHRESHOLDCLIMB 10  // Threshold at which climbing is indicated cm/s
  #define VARIOTHRESHOLDSINK -20  // Threshold at which sinking is indicated cm/s
  #define VARIOHZ    40           // 20cm/s = 1hz --> 200cm/s = 10hz
  #define VARIOMAXCLIMB  200      // Maximum climb rate for maximum hz
  #define VARIOMAXHZ    5         // Maximum hz for max climbrate

Thanks for being tester here. I will build circuit myself. and now I must admit I really want to FPV thermal. Been on the todo list for some time...

ShikOfTheRa, 2017-06-22:
Just tried some debug and as far as I can see it looks like its working on a debug screen - but with out speaker setup its hard to tell. I made a small tweak to improve screen switching.
Hard to visualize how it sounds, but I hope now its just tweaking frequencies, hz and trigger limits. Probably better to test with a barometer. Think GPS will work better in the air than ground.

Arduino tone looks quite heavy on memory so once we have this working I'll probably do some dedicated code. 

Cool improvement if we get this working well

Schalonsus, 2017-06-22:
Second test:
```
  #define VARIOFREQCLIMB 1500      // Climb tone Hz
  #define VARIOFREQSINK 400       // Sink tone Hz
  #define VARIOTHRESHOLDCLIMB 10  // Threshold at which climbing is indicated cm/s
  #define VARIOTHRESHOLDSINK -20  // Threshold at which sinking is indicated cm/s
  #define VARIOHZ    20           // 20cm/s = 1hz --> 200cm/s = 10hz

  #define VARIOMAXCLIMB  300      // Maximum climb rate for maximum hz
  #define VARIOMAXHZ    15        // Maximum hz for max climbrate
  #define VARIOMINHZ    2         // Minimum hz for min climbrate
```
https://youtu.be/a-V_Q4d8Sog

FPV thermalling is awesome, here a video of my 140g MiniSolius:
https://youtu.be/azhiioGl65M

Schalonsus, 2017-06-23:
Would be cool if we can tune the sounds similar to the ones from the KK Vario, really like this style of tones.
https://www.youtube.com/watch?v=A7DH6d09TaA

ShikOfTheRa, 2017-06-23:
ah dont post videos - I really don't need another plane!! its not helping me to resist temptation.

For vario - well that's a pretty good start. Thanks for vid - helps a lot as usual. These are my thoughts:

So for HZ I have made an error there - it is half what expected so will correct that.

The change in frequency is audibly less than expected so think will add a scale in there to allow tweaking. Will make it sound a bit more sensitive.

I listened to a few more varios on youtube last night and some have just beeps other have a varying tone around "approaching" sink / climb. Might be an option to add.

For testing, the GUI doesn't send vario changes very often. In real MWOSD use it will match GPS hz for GPSOSD operation and be twice a second for vario data from inav type FC. We can increase that a little for better performance. Will have a think about that.

I'll see if I can make up a test rig.. Would be better to develop these ad give you something working first time!

ShikOfTheRa, 2017-06-23:
Ok - have made a test rig. Well earpice plugged directly onto rssi pin with 500 ohm resistor seems to work.

Have read up a little on varios and and i think we can get this much closer to a real one sound wise. I guess they have evolved over time and the frequency / tone gaps are there after lots of testig and feedback.
I'll work on this next week Going away today for a couple of days but like to finish this off next week.

Schalonsus, 2017-06-23:
Awesome, looking forward to test the improved version. 
I have another plane where i could add a baro via i2c, so testing should be much easier than with the gui.

ShikOfTheRa, 2017-06-26:
OK

Uploaded something to test. Its a bit more like some of the ones on youtube - but there is a lot of variety out there!

Also something to think about...
Thermal assist... remembers location of best thermal in last minute and provides distance / direction to it...
I was dreaming last night and it came into my mind that you can't see thermals, but once in tehm coupld create a map. It would have to be short lived - importance decay over. With good OSD hardware could map multiple!

Let me know what you think.

ShikOfTheRa, 2017-06-27:
I googled a bit more last night whilst looking at MS5611 and found kkvario. probably could have saved some time...

So on my version of the minimosd I broke out I2C pins (4  pins behind serial on pic below)
4 wires to a MS5611 and a bit of code and you have a proper vario!
Could pretty much take the KK code and add it in 30 mins...

Ordered and will let you know how it goes.

[image]

Schalonsus, 2017-06-27:
KK Vario is the one i use successfully in all my big planes.
The idea was to use the present hardware (OSD + FC) to create a simple (just use GPS if no baro connected or not available on FC board) vario for small, very light weight planes.
On the most OSD there are no I2C broken out.

I could just take a Arduino Lilypad and a MS5611 and have a tiny vario, but that would be additional weight and space i dont have on my tiny planes. So the idea was to use present hardware.

If you could make KK Vario code work with the OSD would be fantastic, love the sounds.
Just the question if this will work with GPS only too?
On my tiniest plane i have a brushed F3 FC to control my servos and brushless ESC, but it has also no I2C broken out where i could connect a baro.
On other planes i have normal STM32 boards that have baro onboard or have i2c. So no need to connect baro to OSD directly.

ShikOfTheRa, 2017-06-27:
GPS only should work. I recall adding in a function to create a GPS vario. If you can use 10hz GPS then should be not too bad. That said - I haven't tested it..

KK vario in code is easy. I am really just tempted to add it in as it is and credit Ralf for his great work. No point in re-inventing the wheel! I like doing that but have a bunch of other things to do first.

https://github.com/ShikOfTheRa/scarab-osd/wiki/Audio-Vario

ShikOfTheRa, 2017-07-11:
@Schalonsus I have added MS5611 to OSD hardware and added kk code. It works!!
Optional extras:
Deadband
Disable audio tones on throttle
Enable / disable audio tones with screen layout switching.

If you want to try, I will send you one of theses FOC to help me in testing / improving. It has I2C broken out so you can add your own. I will do some documentation for it: 
https://www.virtualpilot.co.uk/index.php?route=product/product&product_id=51

ShikOfTheRa, 2017-07-11:
Send me pm on any of the forums.

ShikOfTheRa, 2017-07-17:
Hi - attached is pinout of the connections for I2C MS5611 module to the 
Aeromax hardware for when it arrives.

At moment, code only seems to work if using RSSI pin as the audio 
output. I have a spare pin unused which was added for future expansion 
on the end but for some reason I can't get it to function. It would be 
great if could have RSSI, Current sensor and vario output at same time.

Cheers,
G

Schalonsus, 2017-07-20:
A few questions:
-Do i need to use the MS5611 on OSD I2C port to get KK vario working?
-Or does it also work if FC has baro onboard?
-If so, does it matter what baro (MS5611/BMP280/BMP085) is onboard?

Schalonsus, 2017-07-20:
Wanted to test new code (latest master) but Arduino IDE says sketch is too big

Schalonsus, 2017-07-20:
I found a config that lets me compile, but i'm very limited regarding settings. Had to disable a lot of features i normally use.

ShikOfTheRa, 2017-07-20:
We can improve on that. First intent is to get it working then I can look to optimise the code. 
Can you let me know what config you require. See what can do :)

Schalonsus, 2017-07-20:
The config i have to use atm:
`/*--------------------------       MANDATORY configurable parameters      ----------------------------------------------------*/
/*--------------------------       MANDATORY configurable parameters      ----------------------------------------------------*/

/********************       OSD HARDWARE settings      *********************/
//Choose ONLY ONE option:
//#define MINIMOSD                  // Uncomment this if using standard MINIMOSD hardware (default)
//#define MICROMINIMOSD             // Uncomment this if using the MICRO MINIMOSD hardware
#define AEROMAX                   // Uncomment this if using MWOSD AEROMAX hardware
//#define RTFQV1                    // Uncomment this if using standard RTFQ/Witespy V1.1 OSD, select this to correct for both swapped bat1/bat 2 and to also use alternative resistors / pinouts.  
//#define RTFQMICRO                 // Uncomment this if using micro RTFQ/Witespy Micro Minim OSD, select this to correct for swapped bat1/bat 2.  
//#define RUSHDUINO                 // Uncomment this if using Rushduino
//#define KYLIN250PDB               // Uncomment this if using a Kylin 250 FPV PDB (Using A6 as VOLTAGEPIN)
//#define AIRBOTMICRO               // Uncomment this if using an airbot MicroOSD
//#define ANDROMEDA                 // Uncomment this if using an Andromeda (http://www.multiwiicopter.com/)

// NOTE-some boards have swapped bat1/bat2 pins and alternative voltage measuring resistors
// If having difficulties, first select default MINIMOSD as above, then use the following to correct: 
// #define SWAPVOLTAGEPINS          // For boards with batt voltage appearing on vid voltage
// #define ALTERNATEDIVIDERS        // For boards with voltage unable to be adjusted high enough

/********************       CONTROLLER SOFTWARE      *********************/
// Choose ONLY ONE option:-
// Note - choose carefully to ensure correct settings are written to flight controller.
// The first three are for convenience - they set the OSD for the latest FC version. 
// IMPORTANT - remember to update MWOSD when updating FC software!!

// Choose ONLY ONE option from the following long list :-

// latest release...
//#define BETAFLIGHT                // Uncomment this if you are using latest BETAFLIGHT version 3.1 onwards
//#define CLEANFLIGHT               // Uncomment this if you are using latest CLEANFLIGHT version from repository (2.0 at time of this MWOSD release)
#define iNAV                      // Uncomment this if you are using latest iNAV version from repository (1.01 at time of this MWOSD release)
//#define APM                       // Uncomment this if you are using APM MAVLINK 1.0 compatible FC
//#define PX4                       // Uncomment this if you are using PIXHAWK with PX4 stack
//#define BASEFLIGHT                // Uncomment this if you are using latest BASEFLIGHT version from repository (Stable 2015.08.27 at time of this MWOSD release)
//#define MULTIWII                  // Uncomment this if you are using latest 2.4 MULTIWII
//#define MAHOWII                   // Uncomment this if you are using MAHOWII (https://github.com/mahowik/mahowii)
//#define KISS                      // Uncomment this if you are using KISS FC
//#define DRONIN                    // Uncomment this if you are using the latest DRONIN MSP Module
//#define NAZA                      // Uncomment this if you are using NAZA flight controller
//#define LIBREPILOT                // Uncomment this if you are using the latest LibrePilot MSP Module
//#define TAULABS                   // Uncomment this if you are using the latest Tau Labs MSP Module
//#define FIXEDWING_BF              // Uncomment this if you are using fixed wing Baseflight 
//#define FIXEDWING_BF_SERVO        // Uncomment this if you are using fixed wing Baseflight with additional SERVO adjustment menu.
//#define HARAKIRI                  // Uncomment this if you are using HARAKIRI (for BOXNAMES compatibility)
//#define RACEFLIGHT                // Uncomment this if you are using RACEFLIGHT - untested. Test and feedback required
//#define SKYTRACK                  // Under development
//#define GPSOSD_UBLOX              // Uncomment this if you are using a UBLOX GPS module for a GPS based OSD
//#define GPSOSD_NMEA               // Uncomment this if you are using a NMEA compatible GPS module for a GPS based OSD
//#define GPSOSD_MTK                // Uncomment this if you are using a MTK module for a GPS based OSD
//#define NOCONTROLLER              // Uncomment this if you have nothing connected to the serial port - no controller or GPS module
// old releases supported...
//#define MULTIWII_V23              // Uncomment this if you are using MW versions 2.2/2.3  
//#define MULTIWII_V21              // Uncomment this if you are using MW versions 2.0/2.1  (for BOXNAMES compatibility)
//#define BASEFLIGHT20150327        // Uncomment this if you are using BASEFLIGHT up to and including version Stable 2015.03.27
//#define CLEANFLIGHT190            // Uncomment this if you are using CLEANFLIGHT versions 1.9 through 1.14.2
//#define CLEANFLIGHT172            // Uncomment this if you are using CLEANFLIGHT versions up to and including 1.7.2
//#define CLEANFLIGHT180            // Uncomment this if you are using CLEANFLIGHT versions 1.8.0 & 1.8.1 
//#define BETAFLIGHT3               // Uncomment this if you are using BETAFLIGHT versions prior to 3.1

[... cut ...]
#define AMPCALLOW  0                     
#define AMPCALHIGH 1023                  

/********************  Advanced parameters  settings      *********************/
// This is to enable rarely used advanced parameter saving. Off by default to minimise risk 
// Enabling may create ability to write invalid data to FC
//#define ADVANCEDSAVE                   // Enables saving of advanced paramters where supported (Looptime/profile)
//#define FIXEDLOOP                        // Enables 1ms loop max for consistency

`

Schalonsus, 2017-07-20:
Thats what i want:

```
#define iNAV
#define FIXEDWING     
#define HORIZON
#define FILTER_AVG
#define ALARM_SATS 
#define ALARM_GPS 5
#define ALARM_MSP 3
#define USEBAROALTITUDE
#define USE_VSYNC
#define DECIMAL '.'
#define ALT_CENTER
#define FORCECROSSHAIR
#define MINSUMMARY
#define NOTHROTTLESPACE
#define APINDICATOR
#define DISPLAYEFFICIENCY 
#define DISPLAYVARIO
#define KKAUDIOVARIO A3 
#define AUDIOVARIORC 1200
#define BATTERYICONVOLTS
#define SHOW_TOTAL_DISTANCE
```

ShikOfTheRa, 2017-07-20:
OK thanks. I'll work to see what can do. Thats a fair chunk of memory. Great to see someone making most of the features
I think I can reduce the vario requirements significant and have a go at some optimisation.

Suggest to disable soem of these if not used. Reasonably big hitters for memory
#define DEBUGMENU                   // Enable to display debug values in OSD menu 
#define SBDIRECTION     // Enable/disable sidebar indicators (changes in speed or altitude)
#define MAPMODE         // Enable/disable MAP MODE - map indication of relative positions of home and aircraft
#define INTRO_MENU                  // Enable to display TX stick MENU 
#define INTRO_CALLSIGN              // Enable to display callsign at startup
#define INTRO_SIGNALTYPE            // Enable to display video type at startup
#define AUTOCAM                     // Disable if no screen display. Enables autodetect Camera type PAL/NTSC. Overrides GUI/OSD settings.
#define DISPLAY_PR                  // Display pitch / roll angles. Requires relevant layout ppositions to be enabled

Schalonsus, 2017-08-14:
The KK code works good as expected.
Only bug i found so far: when gliding with no throttle and then applying throttle, the vario is stuck on the sound it generated at this moment

ShikOfTheRa, 2017-08-14:
Awesome!! That's not bad for first test. 

until I update, you can insert a noTone like below...
  if (MwRcData[THROTTLESTICK]> AUDIOVARIORC) {
    noTone(KKAUDIOVARIO);
    return; 
  }

I have started on reducing memory to allow this to co-exist with other features. Shouldn't be any problems to do this. Maybe in future make some of the options configurable on the gui

ShikOfTheRa, 2017-08-16:
Have updated main repo with this.

Also do you think the defaults for  deadband, throttle climb etc. are OK?

Schalonsus, 2017-08-16:
Thanks.
Had just one flight so far, but i think deadbandlow could be a bit smaller.
If i have time for flashing new code i can test today.

Schalonsus, 2017-08-17:
Set deadbandlow to -25
https://youtu.be/gUqsni0WE1A

ShikOfTheRa, 2017-08-17:
Damn that is awesome. Fantastic setup.
FPV glider is 100% on my radar for a build this winter.

ShikOfTheRa, 2018-01-11:
@Schalonsus just as fyi, in next release, will be able to accommodate  all the items requested as long as you don't mind losing map mode...
Also will gain a few items..
1 - The vario slider will be replaced with up/ down arrows, but can be a single character or up to full display height of OSD display for maximum resolution. like here https://www.youtube.com/watch?v=oHV9FDzLrHQ
2 - Vertical speed numeric indicator is back after long time without.

#define iNAV
#define FIXEDWING     
#define HORIZON
#define FILTER_AVG
#define ALARM_SATS 
#define ALARM_GPS 5
#define ALARM_MSP 3
#define USEBAROALTITUDE
#define USE_VSYNC
#define DECIMAL '.'
#define ALT_CENTER
#define FORCECROSSHAIR
#define MINSUMMARY
#define NOTHROTTLESPACE
#define APINDICATOR
#define DISPLAYEFFICIENCY 
#define DISPLAYVARIO
#define KKAUDIOVARIO A3 
#define AUDIOVARIORC 1200
#define BATTERYICONVOLTS
#define SHOW_TOTAL_DISTANCE

I'll close down for now. Feel free to re-open or just comment as I still see comments after closing

closed 2018-01-11 by ShikOfTheRa