ShikOfTheRa/scarab-osd #435: charset problem, osd on apm 2.6

turborocket, 2017-06-18:
Hi all,

i have 2 micro minimosd, both with 1.7 fw and bth with same config.

But 1 of those 2 has not clear on screen character.

[image]
[image]

as you can see in the picture, the hud is customized, but the character are not clear, specialy when the video out is on.

in the photos, i was using an lcd display with a external vrx, but the same problem is on my goggles.
There is a picture of wiring
[image]

some advice?

ShikOfTheRa, 2017-06-19:
There was a bug with one of the GUI versions which created an effect just like that.
I think to resolve, if you try reload fonts with latest it should be OK.

Another option is to use font uploader from utilities in flasher

ShikOfTheRa, 2017-06-27:
Any luck?

turborocket, 2017-06-27:
hello there! thx for the tip, i didnt tryed yet becouse im full of work in the last days!

i hope my son will sleep not too late tonight, so i can try it ;)

what are you suggesting me? i used the last 1.7 gui (payed), and what is the font uploader rom utilities in flasher?

ShikOfTheRa, 2017-07-04:
You can use fontloader rom - or the font uploader button in GUI. Hopefully it will work. There was a definite issue with GUI. It should have updated automatically in background now. Please try again with 1.7.1.5

[image]

turborocket, 2017-07-18:
Hello,

i solved the problem by upload button on the 1.7.1.5 GUI

Thx a lot for the support!

closed 2017-10-30 by ShikOfTheRa