ShikOfTheRa/scarab-osd #436: PX4 Compatibility

4cellOstrich, 2017-06-19:
Hi all,

I have been working for awhile on a fully compatible osd using minimosd hardware with px4 pixhawk fc's. I, unfortunately, have made very little progress. The issue lies in that the flight mode architecture is vastly different between the supported apm and unsupported px4, as well as some other data loss that I have not found the source of it yet, if anyone has a method that has fixed this, I would be extremely grateful! I really am an incompetent programmer, so I probably have overlooked some things.

Thank you!

ShikOfTheRa, 2017-06-20:
I haven't used PX4 so can't help you unfortunately. I take it that it doesn't use mavlink? Are there any links that describe the protocol. 
Is this to add to MWOSD - more to create your own?

4cellOstrich, 2017-06-20:
Well it does use mavlink, however the structure is somewhat different, attitude data is conferred, but flightmodes, satellites, all that jazz is lost. And yes, ideally my goal is to make a compatible firmware. I can send you to the night-ghost minimosd repository where I tried, with no luck, to accomplish the same goal.

ShikOfTheRa, 2017-06-21:
Hi - do you have a tlog of px4 serial data?

4cellOstrich, 2017-06-30:
Unfortunately I do not.

ShikOfTheRa, 2017-07-18:
Hi - I started to look. At first glance see two areas
1 RC Channels / RSSI use different structure - I have added this in ready to test
2 Flight Modes - this looks to be different. Not quite clear tom me if its an issue or not.

The rest of the info - GPS etc. all looks normal - as long as its sent it should be displayed.

https://github.com/ShikOfTheRa/scarab-osd/tree/PX4

ShikOfTheRa, 2017-07-19:
@4cellOstrich I have now added support for the different PX4 modes.
Just needs testing now to see what works / doesn't

4cellOstrich, 2017-07-19:
Amazing! Ill test ASAP!

4cellOstrich, 2017-07-19:
Worked like a charm! Many, many, thanks!

ShikOfTheRa, 2017-07-20:
OK - pleasantly surprised. Let me know if anything is not working correctly

4cellOstrich, 2017-07-20:
so i guess there is one thing, even though the features seem to work perfectly, it has no longer interfaces with the configurator, which is so far the only issue i can see.

4cellOstrich, 2017-07-20:
also, the artificial horizion movement is inverted on the pitch axis

ShikOfTheRa, 2017-07-20:
I had two others on APM to say pitch reversed so I have just amended. Surprised hasn't come up more often.

All updates are now in main repository. PX4 test branch removed.

For configurator - its working OK for me - have you set baud rate to match OSD - 56k default for PX4 ?

4cellOstrich, 2017-07-20:
OK yeah i had the baud rate wrong LOL

ShikOfTheRa, 2017-07-20:
Good guess - Only because I do it all the time!!!

4cellOstrich, 2017-07-20:
This will sound asinine but in the new master release what hex file is the PX4 compatible one?

4cellOstrich, 2017-07-20:
and is it available on the config tool auto download?

4cellOstrich, 2017-07-20:
nvm! it works! if there are any more issues, ill be sure to let you know! Thank you so much man!

ShikOfTheRa, 2017-07-20:
Hex files not there yet for download so only by using Arduino at momnet. I'll be working on a way to get test releases out quicker or direct from GUI config.

closed 2017-07-20 by 4cellOstrich