ShikOfTheRa/scarab-osd #438: Current indicator is not working witn cc3d inav and cicro minim osd

retif, 2017-06-22:
Usinc cc3d with iNAV 1.6.1. And micro minim osd with mw osd 1.7.

If autopilot attached current indicator is zero all the time. If autopilot deattached - current indicator works as expected.

ShikOfTheRa, 2017-06-23:
Where is current connector connected?  To FC or to OSD direct.
And on GUI Amperage tab can you paste a screenshot in here?

retif, 2017-06-23:
[image]

Current connector connected direct to osd

ShikOfTheRa, 2017-06-23:
Ill have to replicate and test this. Will be next week unfortunately as away until then.

ShikOfTheRa, 2017-06-27:
I tried with inav and naze controller.
Seems to work OK for me.

Note – if disconnected, the amperage pin input will “float”. 
As you are seeing a voltage with it disconnected that indicates that firmware is working.

Is it possible to connect a voltmeter to the OSD current sensor input so see if any voltage being registered? Should see it rise as apply throttle

ShikOfTheRa, 2017-06-29:
Mmm might be an issue. A different setup to yours but running current through FC was not displaying on OSD.

ShikOfTheRa, 2017-07-11:
I have been away, but I can say I seem to replicate issue in plane.

ShikOfTheRa, 2017-07-11:
I hope to test and verify tonight.

ShikOfTheRa, 2017-07-11:
Fixed in main repository. 
It occurs if only use FC amperage. If use FC volts or FC RSSI it works OK. I think from memory it was by design assumed if using FC amperage you would always use volts.

closed 2017-07-24 by ShikOfTheRa