ShikOfTheRa/scarab-osd #439: "Sketch too big" Arduino with Arduino pro or pro mini

mawright89, 2017-06-23:
I cant compile the sketch for any of the versions of MWOSD in arduino 1.8.2 with arduino pro or pro mini. I always get a failure that the sketch is too large.

Im trying to flash an onboard OSD on an emax Skyline32 OSD advanced which i believe needs the MinimOSD settings in MWOSD but i could be wrong so advice on that would be good too.

any help apreciated.

ShikOfTheRa, 2017-06-23:
You have wrong board selected. Not 168.   It's 328 which has double memory :)

mawright89, 2017-06-24:
Haha omg what a bell end I am.. Thanks, appreciate the help. Can't believe I missed that.

mawright89, 2017-06-24:
So now another issue... compiled correctly however.... tries to upload data however the light on the FDTI board blinks for around 5 mins then I get the error below. "an error occoured while uploading the sketch."

Arduino: 1.8.3 (Windows 10), Board: "Arduino Pro or Pro Mini, ATmega328 (5V, 16 MHz)"

Sketch uses 23332 bytes (75%) of program storage space. Maximum is 30720 bytes.
Global variables use 1895 bytes (92%) of dynamic memory, leaving 153 bytes for local variables. Maximum is 2048 bytes.
Low memory available, stability problems may occur.
An error occurred while uploading the sketch

ShikOfTheRa, 2017-06-26:
Coulb many reaosns for thsi.
I'd start with 
1 correct com ports selected
2 is cabling correct between FTDI and OSD (is tehre is any cable...) must be 6 pins and correct rx>tx
3 google!

mawright89, 2017-06-26:
ahh, right my FTDI cable only has 4 ports, 5V, GND TX and RX. What FTDI adapter would you recomend?

Many thanks for the help with this. I really appreciate it.

ShikOfTheRa, 2017-06-26:
Thanks for update

closed 2017-06-27 by ShikOfTheRa