ShikOfTheRa/scarab-osd #443: Amps zero adjust unable to use negative numbers.

TronskiFPV, 2017-07-05:
Using negative numbers would be helpful for current sensors that start at 0V like the Attopilot series. Using positive numbers on these sensors results in 0A being shown while amps are actually being drawn. Because of this you can only use "amps adjust" and must make a compromise of accuracy on the low end for accuracy on the high end. 
This video shows how to use both amps zero adjust and amps adjust together to get the best accuracy across the range. 
https://www.youtube.com/watch?v=AWjblvHgjjI

ShikOfTheRa, 2017-07-07:
Am away on cellphone only. I think in config.h there is an option for those that need a negative offset.

TronskiFPV, 2017-07-07:
I cant find anything config.h or in def.h for negative numbers. I have looked around and I cant seem to find where in the code these variables are applied to the calculation for current.  Enjoy your vacation!

ShikOfTheRa, 2017-07-11:
ha you are correct. I was thinking of the old method.

The way MWOSD sensor code works is quite different to betaflight etc. I would like to re-visit at some point. Mainly to optimise memory though as bear in mind this is only old 8 bit OSD not always using floating point so can get some rounding errors.
From experience, current sensor accuracy seems poor at low current drawn

So I am guessing that even with zero adjust =0, the amperage drawn is showing to low? 
I've tried some simulations and it works in all scenario except where sensor still outputs 0v for low amperage drawn.

Added this into config.h which should enable you to tweak it a little.
/********************  AMPERAGE calibration  settings      *********************/
// This is used for those who want to specify non default amperage calibration values. 
// These should only be used if unable to calibrate within limits on GUI. 
#define AMPCALLOW  0                     
#define AMPCALHIGH 1023

TronskiFPV, 2017-07-12:
Awesome! Is there a correlation to the numbers? Do these numbers define the low and high point of the scale from 0v-3.3v or do they act as capped values on the ends? Hopefully I can compile and test tomorrow 👍

ShikOfTheRa, 2017-07-12:
Notes – 
almost always should set analog reference to 5v if using a current sensor. If it will output more than 1.1v it is mandatory. 
Current sensors are not always inaccurate.  
Meters for measuring are not always accurate. I have 2 tested last night with 1A difference at 15A 
RC gear draws current as well as motor so should never see 0A. Mine draws 0.2-0.3A at idle

The code:
Maps the Arduino analog inputs (step values of 0-1023 steps between 0 and 5v Vcc)   from zero amps to the current drawn if the sensor could go up as high as 5v. 

So for an example 90Amp current sensor with a sensitivity of 36.6mV per Amp drawn and a zero Amp offset of 20mv :

For Zero Amp calibration:
“ZERO amp adjust” = 1024 * 0.020A / 5v  (range * offset / maximum voltage) = 4

For MAX Amp calibration:
“Amps adjust” = 5v / 0.0366A  (Max voltage Arduino can read / sensor voltage per amp drawn) *10  = 1366 Because we require accuracy to 0.1A we use 1366 as amps adjust number

So for your tweak, it could be that the sensor has a “negative offset” – so 0v sensor out = 0.5A draw. By changing #define AMPCALLOW  to -5 this would accommodate.

Once “Amps adjust” is roughly set, generally reduce the  ZERO amp adjust until it records correct amperage.

TronskiFPV, 2017-07-14:
the new variables dont appear to be working. I even tried AMPCALLOW -100 just to see if I could get it to show some amps at idle, but it still shows 0 and seems to work as it always did.  

I'm not shooting for extreme accuracy, just enough to use MAH consumed as a rough indicator of battery left for long distance flights. I could have other issues here, but with some more room for calibration I should see some improvement.  I've ruled out a bad 5v reference because I can power the board with USB/FTDI and get the same results.  The sensor outputting 0-3.3  instead of the max 0-5v dosnt help anything of course.  

Where do these variables all come together in the code to form the solution displayed? I looked again and dont understand where that is. I'd like to mess with it if I can figure it out.  Hell I'd imagine I could write a table for this particular current sensor given that lol.

closed 2017-10-31 by ShikOfTheRa