ShikOfTheRa/scarab-osd #448: Betaflight3.1 no data on MW 1.7.2

4cellOstrich, 2017-07-30:
Hate to bother again, but,

I just updated my Kombini from BF 3.0 to 3.1, as well as my sp3 Vtx/OSD from mystery 1.6 to BF MW 1.7.2, however there is no data anywhere, all zeros. After looking into it I found that there was the issue with MSP, but that was addressed in this firmware so I am a little lost. I flashed from Arduino, tried to use LTM instead of msp, but just no luck anywhere. still the same stupid 0v 0%rssi and 0A staring me in the face. 

Sorry to bother again!

ShikOfTheRa, 2017-07-31:
is MSP enabled at correct speed on correct port?  I have had no other reports

4cellOstrich, 2017-07-31:
Uart1 at 115000?

4cellOstrich, 2017-07-31:
i am about 90% sure that the port that im connected to is uart 1

ShikOfTheRa, 2017-07-31:
OK as long as GUI not plugged in at same time. Will have to unlpug OSD each time want to use GUI.
I assume both tx and rx lines are connected

ShikOfTheRa, 2017-07-31:
Enable hem all and see what happens!

4cellOstrich, 2017-07-31:
no the tx is unplugged from usb when its powered, i guess ill do that and see what happens!

4cellOstrich, 2017-07-31:
well i enabled them all and still nothing. This board was definitely working before so it has to be software...

ShikOfTheRa, 2017-07-31:
I can test in a couple of days.

4cellOstrich, 2017-07-31:
awesome, ill keep you posted if anything changes

ShikOfTheRa, 2017-08-02:
I just connect up betaflight 3.1 and 3.2 to MWOSD 1.7.2 and at least some of the functionality works.
I have no RC available to test further:, but the following at least work OK
Artificial horizon
Baro 
main voltage
rssi

4cellOstrich, 2017-08-02:
Nuts. What was fc/osd combo?

ShikOfTheRa, 2017-08-02:
FC is a RMRC dodo clone and the osd is a stock minim.

4cellOstrich, 2017-08-02:
damn, ill try again, but im pretty sure its not a hardware issue bc its on 2 quads, do you have any special port setup?

ShikOfTheRa, 2017-08-02:
Nope. Just a default setup then enable MSP on port and ensure its at correct baud rate

4cellOstrich, 2017-08-02:
which is 115000?

4cellOstrich, 2017-08-02:
i just dont get  it, it doesnt recognise arming either

4cellOstrich, 2017-08-02:
did you just flash the osd with betaflight MWosd and it worked? using the configurator?

ShikOfTheRa, 2017-08-02:
Yep. Osd - betaflight minim 1.7.2 from configurator 
For FC I uploaded latest hex from releases. Think it was 3.2 rc x

4cellOstrich, 2017-08-02:
i guess ill go to 3.2 and try it? ive heard good things about it.

4cellOstrich, 2017-08-02:
actually im seeing that BF 3.1 in general was pretty weird on a lot of things so i hope 3.2 will be stable

4cellOstrich, 2017-08-09:
yeah switching to 3.2 fixed it, so, who knows!

closed 2017-08-09 by 4cellOstrich

ShikOfTheRa, 2017-08-09:
Thanks for feedback. Brill