ShikOfTheRa/scarab-osd #449: Flight Mode for Pixhawk

ajtank, 2017-08-13:
I use a Pixhawk Flight Controller.
If I load PX4 v1.7.1 or v1.7.2, Flight Mode is missing and is replaced by a text 'PX4' on the screen.
If I load APM v1.7.2, Flight Mode appears and responds to the FC.
This problem does not exist in v1.6.8 Pixhawk firmware.

ShikOfTheRa, 2017-08-14:
Hi – in older versions there was no difference between Pixhawk and APM.

My PIXHAWK/PX4/APM knowledge has been non existent until recently. So now I understand that I understand now that for Pixhawk you can install APM stack or PX4 firmware stack.
In the latest version of MWOSD, you need to matche what is on your FC with OSD by installing PX4 or APM

“PX4” for mode means it’s not receiving valid PX4 modes. It probably means you are running APM stack on the FC

“This problem does not exist in v1.6.8 Pixhawk firmware.” – you are correct. This is because in older versions it was actually running APM code on MWOSD

To me it sounds like you are running APM on your PXHAWK.

ajtank, 2017-08-15:
I see. Indeed I am running APM on my pixhawk. No wonder why in the config.h I see wordings of 'PX4 stack' for pixhawk.

Thank you for the explanation. Problem resolved.

closed 2017-08-15 by ajtank

ShikOfTheRa, 2017-08-15:
No worries - your not the first to ask. Or to try to figure it out. I might amend the config. h to just say APM / PX4. remove the reference to pixhawk altogether because that's just the hardware platform.

Thanks for the feedback!