ShikOfTheRa/scarab-osd #450: Chrome configurator will not connect

adamzimnyy, 2017-08-20:
Hello. 

I flashed the 1.7 version onto micro minimosd and was trying to connect using chrome app, but nothing happens after I click Connect. The log just shows "Connecting" and nothing else happens. 
I downloaded everything I use today, so I believe the version should be correct.

ShikOfTheRa, 2017-08-21:
Are you using correct baud rate? 115k for MSP systems like xxxflight and 56k for APM/PX4

Also - try changing baud rate to a different value then resetting it back.

zsoltm, 2017-08-22:
Same here.... Win10 / Chrome. If I flash 1.6 and use the old java client, it works. Other chrome apps, like Betaflight could connect to a FC on the same PC.

yds, 2017-08-28:
@ShikOfTheRa 
> try changing baud rate to a different value then resetting it back.

That worked! Thanks for the tip. I've been having the same problem with [R1.7.2.2 Test Release](https://github.com/ShikOfTheRa/scarab-osd/releases/tag/1.7.2.2).

adamzimnyy, 2017-08-28:
I tried changing baudrates, but still can't connect. Anything else I can try?

ShikOfTheRa, 2017-08-31:
If you use pc and send email to spam<at>flypix.co.uk I can send you a PC client to try instead.

It is surprising that it flashes OK, but unable to connect for normal configuration... very starnge

slawekgajzner, 2017-09-04:
Hello.
I also can not connect to version 1.7.

If I use the configurator from 1.6 it connects but says that the configurator is not up to this version.

flexbex, 2017-09-07:
What worked for me was touching serial port and baud rate selector one time. Because values are shown but not selected internaly.

slawekgajzner, 2017-09-07:
Thanks, it works.

asdiasx, 2017-09-12:
Had this problem and solved chanding baudrate for the com port in windows... it was 9600 by default...

ShikOfTheRa, 2017-10-30:
This is now included in latest GUI configurator

closed 2017-10-30 by ShikOfTheRa