ShikOfTheRa/scarab-osd #455: fixed wing has no win speed and trip distance

wetouns, 2017-09-22:
I am fixed wing player,so i need win speed and trip distance data,Can you add these feature to the next version?

ShikOfTheRa, 2017-09-26:
Hi - do you mean wind speed? If so yes this is intended. Likely to be around January for next release.

Trip distance - Summary should have trip distance.
For always on trip distance, this can be added, but only by  compiling and uploading with Arduino  :
/********************  RECORD CAPTURE  settings      *********************/
//#define SHOW_MAX_DISTANCE              // Enable to display MAX distance achieved on line below current distance

It may be in future version if there is enough memory

wetouns, 2017-09-29:
tks,I found how add trip distance and airspeed,and I modified your code to support airspeed and ground velocity show bouth . my english is bad,in fact I mean airspeed,not wind speed.

macfly1202, 2017-09-29:
For all Ardupilot / px4 long range user on flying wing, Airspeed is one of the most important metric to have especially for efficiency flight.
@wetouns : could you share the parameter you add ?  thanks !

ShikOfTheRa, 2017-09-29:
Yeah until recently I think no-one really used mwosd for px4/ardu - and the main multiwii based ones didn't have support for it.
Next release need to add it in... Both airspeed and wind.

macfly1202, 2017-09-29:
Since we now have PX4 as a 'ready to use' firmware between Ardupilot and iNav, people will start to test it and forgot all "pain" minimosd R800, night-ghost and so on that no one maintains anymore...? Will be in this party soon to fill issues if needed  ... Thanks !

wetouns, 2017-09-29:
@macfly1202 

find screen.ino and replace function "displayGPS_speed(void)" with the code below and build

`void displayGPS_speed(void)
{
  if(!GPS_fix) return;

// #ifdef USEAIRSPEED
//  GPS_speed=AIR_speed;
// #endif
  
  if(!armed) GPS_speed=0;
  uint16_t xx;
  uint16_t air_xx;//Air speed
  if(!Settings[S_UNITSYSTEM]){
    xx = GPS_speed * 0.036;           // From MWii cm/sec to Km/h
    air_xx = AIR_speed * 0.036;
  }else{
    xx = GPS_speed * 0.02236932;      // (0.036*0.62137)  From MWii cm/sec to mph
    air_xx = AIR_speed * 0.02236932;
  }
  if(xx > (speedMAX+20)) // simple GPS glitch limit filter
    speedMAX+=20; 
  else if(xx > speedMAX)
    speedMAX=xx; 
  if(!fieldIsVisible(speedPosition))
    return;
  if (Settings[S_SPEED_ALARM]>0){
    if((xx>Settings[S_SPEED_ALARM])&&(timer.Blink2hz))
      return;
  }    
  screenBuffer[0]=speedUnitAdd[Settings[S_UNITSYSTEM]];
  itoa(xx,screenBuffer+1,10);
  MAX7456_WriteString(screenBuffer,getPosition(speedPosition));
  itoa(air_xx,screenBuffer+1,10);
  MAX7456_WriteString(screenBuffer,getPosition(speedPosition)-LINE);
#ifdef SHOW_MAX_SPEED
  itoa(speedMAX,screenBuffer+1,10);
  MAX7456_WriteString(screenBuffer,getPosition(speedPosition)+LINE);
#endif //SHOW_MAX_SPEED
}`

clemtrinc, 2017-12-16:
Hi, is there a way to add replace that with antoher indication we don't use for fixed wing ?
It would be great to have both airspeed and ground speed

ShikOfTheRa, 2017-12-17:
In new year will be version that has what you are looking for. Aiming for end January.
Any other items available used by APM / PX4 you would like.
I have never used APM, but do now have a tricopter on PX4 for testing but TBH, I'm not familiar with its capabilities yet.

Completed and committed in development repository:-

Wind direction / speed indicator
OSD - PX4/APM – Airspeed
OSD - PX4/APM – Wind direction indicator
OSD - PX4/APM - defaults to text mode for MAVLINK modes
OSD - PX4/APM - key MAVLINK modes supported

APM/PX4 development:-
Windspeed
Waypoint direction
Distance to current Waypoint
Climb/Descent rate and slider guage
Review use of baro / gps alt. GPS alt always used in fixedwing currently

ShikOfTheRa, 2017-12-29:
I am closing tis for now as its been transferred to development roadmap.
All are done pending GUI upgrading to support.
Targetting early feb for release, but there have been a lot of changes in background for next release so will need some extensive testing first.

Can see some of the extra options available here. All can be enabled / disabled from GUI
[image]

closed 2017-12-31 by ShikOfTheRa