ShikOfTheRa/scarab-osd #456: AUTOCELL and LIHV / High Voltage Lipos

ghostface, 2017-09-24:
Heya,

What's the recommended way (if there is one) to get the AUTOCELL/AUTOCELL_ALARM feature to work with regular 3s/4s lipos and high voltage ones?

Regular Lipos go till 4.2V per cell and Lihv to 4.35, thus on regular settings the Lihv ones get detected as empty 4s Lipos.

If I just set CELL_VOLTS_MAX to 44 the battery indicator is off as it will scale based on this.
Or is there no way around this?

Thanks a bunch!

ghostface, 2017-09-24:
Ok since I couldn't find any code to take care of this I monkey patched LiHV detection in, let me know if you are interested.

ShikOfTheRa, 2017-09-26:
What voltage range do they have? Haven't used.

#define AUTOCELL
#define AUTOCELL_ALARM

Set GUI/ OSD menu alarm to say 35 to warn at 3.5v per cell as example

Detecting Cell count is critical:
#define CELL_VOLTS_MIN 34           // Specify the cell voltage at which it is considered empty
#define CELL_VOLTS_MAX 44           // Specify the max normal LIPO cell voltage

Challenge is to make sure that 4*low batteries value is greater than 3 * full batteries.
In reality its not so bad since normally connect only full batteries before flying !! well usually...

closed 2017-12-31 by ShikOfTheRa