ShikOfTheRa/scarab-osd #457: Temperature Sensor - LM35

Schalonsus, 2017-09-26:
Hi,
i connected a LM35 to my OSD for monitoring the temperatur of the motor of my race wing.
+5V, Vout of sensor to RSSI (A3), GND

It shows a value, but it jumps around a lot when i bend the cable of the sensor or squeeze it.

Also changed the code to:
temperature = map (temperature, TEMPZERO, 307, 0 , TEMPMAX);
with TEMPZERO=0 and TEMPMAX=150

So what can i do to improve temperature reading?
Another sensor? Or are there any smoothing/stabilising circuits for LM35?

ShikOfTheRa, 2017-09-28:
I think you are first to try! I have never used myself either.

There is same filtering as on voltage. So to help, maybe try:
Smooth power supply
Not touching cable!!
Could also try different filtering by enabling #define FILTER_HYSTERYSIS 2 

But really there is some quite heavy smoothing 

If using 1.1, suggest using:
 map (temperature, TEMPZERO, 1023, 0 , TEMPMAX);
TEMPMAX=110 (it will read 1023 at max 1.1v

Schalonsus, 2017-09-28:
I think i might have solved the problem.
A 50ohm resistor in line with sensor output and a 100nF film capacitor parallel to sensor output and ground 👍 
Now i need to make sure the sensor has proper connection to the motor case and then test fly.

ShikOfTheRa, 2017-09-28:
Ah jeez - I didn't think of that. Of course.

closed 2017-12-07 by ShikOfTheRa