ShikOfTheRa/scarab-osd #460: MWOSD v1.7.1.9 (windows client) Button Write not Working

vitorzone, 2017-10-25:
I have flashed the OSD, the button Read works but the Button Write it's always inactive.

ShikOfTheRa, 2017-10-25:
Hi - I have just installed it and it is there for me.. Did it register OK?

vitorzone, 2017-10-25:
Hi. Register? I bought the configurator by "https://www.virtualpilot.co.uk". How can I get the code to register? Thanks.

ShikOfTheRa, 2017-10-25:
You should have received an activation code. Drop an email to them requesting resend of key. Will need your email address it was purchased with.

vitorzone, 2017-10-25:
I already ask for the activation code to them. Let's wait now. Thanks.

vitorzone, 2017-10-26:
I did not have that registration information, when I bought the configurator. It's pefect now. Thanks.

ShikOfTheRa, 2017-10-26:
OK - thanks for feedback!

closed 2017-10-26 by ShikOfTheRa