ShikOfTheRa/scarab-osd #461: [feature request] some additional information.

eephyne, 2017-10-27:
Hello.

I use mwosd in my rc car.
A car with an osd ? you say. Yes for long range exploration (at least middle range).
I use it with gps (no flight controller) to get home position mainly but also speed and distance.
About distance, I wonder if this would be hard to add the total distance traveled.
I did'nt look at all inside so I don't know how difficult it would be but in term of calculation, having a int incrementing every time the gps send new coordinates seems easy.

could it be possible ? If you don't have the time to work on it or not interested, could you indicate me a starting point to work on it ?

thanks

ShikOfTheRa, 2017-10-27:
If you upload using arduino instead of precompiled config from the GUI you can enable this in config.h 

#define SHOW_TOTAL_DISTANCE            // Enable to display TOTAL distance achieved on line below current distance

eephyne, 2017-10-27:
oww didn't remember this line, but I already have max distance active, what happen if I activate both ?

ShikOfTheRa, 2017-10-27:
Will overwrite 
Look in screen.ino and search for the the total distance one. Find the bitbin that section that says +LINE and add another +LINE and you will get all three distances. 
If not sure I can post when get on to a pc over the weekend

eephyne, 2017-10-28:
Found it.
Instead of making it static I did this
`#if defined SHOW_TOTAL_DISTANCEÌ                                              
    formatDistance(trip,1,2);                                                   
    #if defined SHOW_MAX_DISTANCE                                               
    MAX7456_WriteString(screenBuffer,LINE+LINE+getPosition(GPS_distanceToHomePo\
sition));                                                                       
    #else                                                                       
    MAX7456_WriteString(screenBuffer,LINE+getPosition(GPS_distanceToHomePositio\
n));                                                                            
    #endif //two lines if max distance is there                                 
  #endif //SHOW_TOTAL_DISTANCE          `

If you think this is clean enough you can put it in the code (I can make a pull request but I think this is a bit overkill for that

ShikOfTheRa, 2017-10-31:
Thank you for the code. I have merged

closed 2017-10-31 by ShikOfTheRa