ShikOfTheRa/scarab-osd #462: Clarification on purpose for the sample 3d dildo file.

MiguelCastillo, 2017-10-27:
Hmmm Hi - Im a bit confused. While browsing the code I came across a rather interesting file in the repo... Do you mind clarifying what the purpose for this file is?  https://github.com/ShikOfTheRa/scarab-osd/blob/master/OTHER/3D/3Dildo.stl

https://github.com/ShikOfTheRa/scarab-osd/commit/1aa20fa30eda8209ece3f1c4c1b1f8e2c6080cf6

eephyne, 2017-10-29:
humm, isn't it obvious ?
Its for a better integration for plugging in some male connector into the board !

More seriously it seem to be part of a test but the file could have been chosen more wisely, that's for sure.
@ShikOfTheRa you're busted !

MiguelCastillo, 2017-10-29:
ok cool. we should add a note in the readme for running these tests. :D

seriously tho. i certainly considered sending a private note to avoid any possible public shaming tone to the conversation. but when i saw the actual commit in the history, the file no longer seemed like a mistake. im happy to close this issue to avoid any unintended consequences.

ShikOfTheRa, 2017-10-29:
Ha, I wondered if anyone would notice!

I printed some out and put them in pockets of random people for fun. 
And some of the guys on facebook wanted the stl so I dropped it up on github. Easiest place at the time!

I probably should remove it, but maybe I'll just leave it there!

MiguelCastillo, 2017-10-29:
for sure - i wanted give you shout to make sure this wasnt a mistake. :)

closed 2017-10-29 by MiguelCastillo