ShikOfTheRa/scarab-osd #466: MinimOSD does not display information when arming drone

n0rf3n, 2017-11-13:
Hi.

I have a problem configuring the MWOSD with the MICRO MINI OSD and flight controller SPF3.

**Problem:**
When initiating the drone I can see the start screen and the information, but when arming the drone the OSD information disappears... and when disarming the drone returns the information to the screen.

_Below I leave several images, the minmOSD, my configuration and a video showing the mentioned error._

I remain attentive to your collaboration 

Hardware => https://www.aliexpress.com/item/32352079419/32352079419.html?shortkey=7nyauIBB&addresstype=600

My Configuration =>  https://drive.google.com/open?id=14zcbQ9pAHXvjD1-gnXs3Efl4lJNiCMg7

**Video Youtube**
[image]

**Images**

Start OSD
![alt text](https://lh3.googleusercontent.com/A9Jb5A4PDS_l29s43pEAbVehEmM7Gp42613BanSzyZ4s1DYE3FaErhnxkcD7sIJeJXQTiPMEinlwAv-vy5_8iiIZsRWYojkV2caT9ZhKuSEhV51WORipjGHNfFNmPkQpQCteyyIT8wRxhoVCPIWCsno4kexB1UuPHDoYjdtBi-fTadAGDH25HGxwBjJL74EOPL5T12SxT7Zvkqkhqek6KU6kuhSsYKmbzODG0rDPogykWezHL5Cs9EE9nZIVTsMlH3L60FWkRc3sOpcoC96QMlD5ZsnBdUZn2Vl-95YPoYOV8ujFi3Cbk-ZzYKuoIIeH5qA2ZcPjp87WQ8aoDqJ2PxIF5tdveV9ZlTO_5dDi1cB-NmOUWJAYyM1CmzrRqPNi2fm4TrqeuhANJvieHeQbooROpxoug2bAwQ3Z3zDHuD29KGejdCr_ArMD-9Gc5U_G7hNx5lm0bfmbGJPdlET6wxeXwqRmml-rAhqrpzMJyLgmmzCA5PhBIFdCauBm0VS5YOKbGUQgXU3vstTKfqKiPBBYe_pJvKxk3PMV-g3y7n395pxKpwQMCHIgfjRIAmQenWEVVbb-4N-blN5GM78ORPeH-LFRE3LwsxVcx2P_BA=w1268-h713-no)
Drone Disarmed
![alt text](https://lh3.googleusercontent.com/oQfVOuOW4tM-wBmB9Hv7znp8pnWjjYu9E0oKvPBRUBvlICCJYAqKLfYKbVc9xKWUXIKIPoeAlL3jseihzysuWeTV-jHBrjY24GYfTHQ7pwGTsZuOifIE2ho0cXdQJ4R7_gq_TVFPUQlaw_0gxU3o1Al4PvJoG7HCJm0OFpakGbXYw2GN_08YQE_ONGQd4AaXnHwgwgBEkc_3kJNN7fhl8cf3WMTjJHJ7fPFO5T5yKEO1U0urJgqwmvaoo9v4zljBrBPaupqiUMwKReq8yELQ2f1nytdLVfNchtFZ16YuGInk5i1QsUBnDFMH-gIIN2zla3MsLr4RWlk_UWQUHIv8nGbt8hB9LAJlU3e97S_klVyDDHbTR_8FKl4V9rbtEFpymfHAIXfEJJwD-hvnpPdA76pD4FJ4RtkLOfbHUaoX885-631ah0y4Q8K9X3DBhvG4Hk5c_ycWtYYG8wgEfwNxJlkkZ6wKDvXfAhf8pXatd3thiXqeYXGLzbfGmjKq3Xl_pQul9ZQ5Ik8Ll2J3fEsQ0NliJ3GbmcdkIBDkcQ7NunqWiJ7M7_Omn20vmHNrdGDRjxk535MpGaAFfgfyvY81_57Y-WdFMEqpGzjUWUz5cA=w1268-h713-no)
Drone Armed
![alt text](https://lh3.googleusercontent.com/HRhBQK8JlIMfVT3GpXL5X9YnUlTlzn2fc41RsnsyRZsvsZqtmIlaVGChgKwIcAyLhd8q0TKwPWQ0fQkR00IqMcxP99TmGZ2rKFSO8EKa5JBYxhZTk0gbZFzlDGMCJA56ZjjziOYNhgayecmM6AtOvHGkAvS11GV7C8Pb5RYFcv-1hK7QyB7o_n_YOFw42Spr68h4dKuW5npGuRHaVtEQWp3Ww5_fntjyZsCwOcllyguEU0r4XPlcrNuYiYzkDdO4kj5Wu7PTWGw7f046OQbLZaTq4QcFBQn-z1uSqwPOQY9_Vq5-Tfvoa6Pw37I_adnaeXjctxeRQ_uZEzji3jmlDK99XSljcI77oTbtqZq88K03U40JS9PBFFa19hJWvRa8AmMP5lhZ4x1fgF3kuQ5Wkfb25nG5yNub4OXTxf7z7988xVH2VJC_hSfBYcjcIPoEEqGY5Km2pVby2nk1TR1Rr55uWEEghJpv_VeMHC_6f--Aoc8SrkTZcUoQ7jilkmFYXakpHAy_n-8iuL-F3fYcCNtIrtQ1zZ3oj1ot-YCRTwsCTXy3I8DM6wmoV3len4XzqokZ2PNG3v2ViKqImsktFpYF5Ut7O1huQxggpb9PLA=w1268-h713-no)

Alejandro Ariza
Thanks And Regards

hobbya, 2017-11-14:
Can you check the supply voltage on the minimosd board before and after you arm the drone? See if there is chance that the voltage drops below 5V.

ShikOfTheRa, 2017-11-14:
The supply is possibly a cause. 

I downloaded your files and installed and its working OK for me.

What version of betaflight are you using? I wonder if something has changed that might impact it.

n0rf3n, 2017-11-15:
Hi @hobbya 

Perform two tests.

Test 1 with power supply from the flight controller via port uart2.
Test 2 with power supply from the PDB.

Result
In both tests the voltage remained stable. 

Here are some images of the test.

## Power supply from flight controller SPF3
**_Drone "Disarmed"_**
![alt text](https://lh3.googleusercontent.com/AeeyE7VEuZbKZqPXB1gngrfqIgzi0xQXUpXhN0UKeFhHvVXZE3K3xagvfsA5_y8YBXBt8IxzRxbyzOtbKSLjX1vXSiS3hINWLaKVY9E5yGe7CLCq1FbJka1E5ePnWVgeGvJY62dfWEbnItFi0JwZAL4z0lPZf3BvnXeAmcr7JdCi5q2eQijRp0jPjIl1KygI7pMMjh1vN-Q2qYF45vSu8KHFnsEdbZKjfTFXm0I5olv3IIYa7ia6-7thS1uobpHKXSkq7MShePYdWxdM4tbvuzk47xVi__zW-n3--P9EHVntYB3bs27gHW-Z2pvrLv92qFrQjEkze2JndaU5pq-95vsYijFHfkLYVvsJ-g5i-iYzCLnMlcWEqllH24IhMBLCo72h5ZSSz1Mgv-lpz9aI5D2P_6mtRzYgE5FcUaOeg3rMs1SrWlyGFpwanMVhssf3e4Ds4WxvBJg1OKD8vH-cU1hMYiCQkDO5krqUUNNGt2xaTf4c9WwkZHBHFatcSOxMAg_2uReB4lc7q4rgNGTJqIE5AR_OqAZwBBrhN1M_uXkZqCLzo93UKHT-D9NU-rDLluhc3xGrJpzga1MG7TGBSFVLPRRTRDe0LCXg2flsXw=w402-h713-no)
**_Drone "Armed"_**
![alt text](https://lh3.googleusercontent.com/rfupcQEE7OrxYPmI0Xp7mhMvqIYzlJ5_I0nwj6qU_dTIaDEtl13fm15mmmTZ7SVLpZZG6Q4tLVOOAPUigXyuruoo4K3nYmq82Q67w-munBWlyN2S6B3W49vobhWdLlO51ArsFj9i0GnKB5p4rPhUYKmljpDxH6e5InHS94wjk3Q0tUB3KrmDMwEewCBi-tvMk2rXYdqN3AV9QIoFa1EHysiCO8odMj6L6heEwyJ6MHmk8uDDgw5ZRzmPX2uEKdmGGaFBu7VMxwTK5eWf4G-DV0mBGWhFYFsr9m8xJGB4_7pZnRL7HL8XdT0Qwk7fz-BDtscoxGV8nWccqvBq0VYhjwoky_wcVYtT_k-lKaLn4qbGutlNlZbdu8lJQL_y6g750KFbuwd2nXLsJLxycVdWWmbyXmLMtpSW99MZ3gdm1qlxhZI656mONo3lvyVGXCpf91VBoZe3GGYZwf-bomWxXXCWOcgXF2reYs07lE2Kil5BGk1g2CL_G9EKrzrMfUmozTvjwQZ0rJVmB8kd5_TEdAL53qXscLq36CT762y4RWl9LizJKY3Ht7uzUbTehN2cj9PKjndqrqrNA4D-vzIOr6VgCPUNIy9wi88Ktv-k_Q=w402-h713-no)

## Power supply from the PDB
**_Drone "Disarmed"_**
![alt text](https://lh3.googleusercontent.com/qV0rZ-2zLfC79heLXk0ds4A-_GIN63f0b7id2tUpMQKQev3hjQVTI6iLy_qgxHHOv05NscC_kih4CVy9O6Ca6tj5c9EqG0LmRca2kdI5Z-EuAb9MfWCHIAGp6gDs0FBaWGFYEfrkujKjMilgKJnYesUvIe-osOvoum75n4uet5oUxwpq1EZ8lsHiLAVG3jabEifwqSbteYeKNn7pK8qa4qyZYPvO66qn78JbGGy75vKzpbDjIF5DkkRAC_eVsDD83DRdf16JkU1A8w4AHCDrkDQf_mbv-SAQAYjRxi9nK2be0FnRIBQ32_iideJVjygmgaqvaKV2oROYV8srBJzmK-A-ZD1sPDeFYr63hfJSaGPdHxR2qhhPx8navbYGCcckWnRN8xQQ4K3LUsQs-1S7f57ttcYLyIgGxhCo5rhtuF5w3ZTan6XUFuFpQyRcpQqNjsQi3bIJSNFnsq3qi0VcDWeo9C4aFy6W1Iy6i1zHCKBtloUm0iHamsHZ5aXuznR-87OgBuKKd0zTyE2Qn2026SVNfSyiGyvBnyn71satcyVGcImdh-E5RdcPiaPo0bpalKLo9Oq-KNt8UFPO-w2SqyskiyHmRRSvKzoQc2MmOg=w402-h713-no)
**_Drone "Armed"_**
![alt text](https://lh3.googleusercontent.com/nb8CErbzDdabWi66kd3lg2Jm39HR_1DCAZ8z56QgOSjgD6RXjd-dypXiPZewjLkZ8VKqclMdtEaD3KqY_1aHg891feyMMwWwNx8u2SDhyFEyZzlwccPGAe0E3SCf6hecYcb2xQiOCGL1paTwFsE3Z21iS1vMvz6NOLRDDgISsgP7VihB8b3wtHRu9AmtyhQdVKTCmg-0ZOdN1AIoRkaegYiCCqSYoRPe5oQY29YaEbp0CJXKIARFxRSp7d941mnuFST-QqkT2RMyYIIpVhoPytsFJQSCzABTUPyxiUNzltRx8fFe2pZW0vztdko141FSzd0U2mPpLV36UE7WK0S4M64NeqgIVwwjYbXFlw_wEUHVopbwVl87imJY-zh80XzT4WX_Ke_78uAh-fH1eCNjyUIjuB4QwfdNWbT3wr1UW4IzapJkyB4p1HUc0AXFN5DyvnFkkZ5RY9g-38INLouggDv5vDL5dJGkJ1a6YC8_yW_lFTbmekNUS9KegE5lPdYmCXURos521-l65dJM_1ADoPNeGz9vLF4OgY7LbIkrJYnYuHtgFBaGd18ilE7LUPjvUp3Cx-L1Z_QnYmPxdcvQW_Rqjh4pEQO_Ka-uxNz5-A=w402-h713-no)

Alejandro Ariza
Thanks And Regards

n0rf3n, 2017-11-15:
Hi @ShikOfTheRa 

Here are some images with the Betaflight version and port settings.

In addition to this I realized that at the moment of assembling and disassembling the drone you can see for an instant the OSD information when the drone is armed, I also leave you with the link to a Youtube video.

**Video Youtube**
[image]

_Version Betaflight_
![alt text](https://lh3.googleusercontent.com/8LDPsNcsmom-DXDToJ-DXKsQV2CzohAsIkYinFlbYN2yNSe9lzG3VWLdfV7waVfCrfGJlWPA0tIPTa6Z8GtrkYFgSwKJtsgX2XA7uM7C-qBvQCvgPVRAcZ9sqtOAaUSsk_uJdoXH0OOE-U5PPW5mTde0-eF90uA2-RbTTLDemQOhkL4nIfqH3zEW-ewMmRGq1I_WGFbjTeGvOuK2pklGTY4YXdX0jmGblnzzccZ15I7fDwypE7gLIHMbx1XJ7xt1IoHYQR93dzuAOMsVY8sMljkeP3hIKuPQTreaWt8NugPffFfAjXqnNnwsriyxmns4_0R86eJ-oPAiX6X58TFJJIYbJkRdcP3ELPbBAkVs0lQq8N2NshCdBlv1TUbwHTNzgc15DkTVVc4TllmR0_8TDT1cjCVWKDKmv4XWCEG0MCqc1uSmDzTicNauL59e9fPWGKm2X_cRAPiXPZ3A5ipWsRyhxRFfwgvQ7vukraRwmgVNJoz_W6aEK9Aih3L3VPvrvKJngKcvfoaHx8oW7tkgnKpgapteAwbmHd-5dniAuNiAh3miiyMnMkusuOLtOc7tzRrTn8vFXPFRzg32HKfRd9l620DFKYzLuqgpi8qX5Q=w847-h287-no)
_Config Ports_
![alt text](https://lh3.googleusercontent.com/xzw82n5BH7sChjdkg7_eBOjs1P1XJy2E-v_L8iOqGlAch9ZP2J3LnomJR5HfLnPpM7hkZ4D0nTqrwPAydEIn9Tk0ke0ZHrYgnf4LgBZEaOm89ijXc4d_MyH-C295mI1YnsGSbjO4epxWjaS2QHF6l-O1yDe0BqMGEPqgQC_RrbuCkstB1VFRk4ZHPGaZg2YTdUp7lTh7ds0cDRRXRF8qSXhZ1Ta0sT0fFkwgfl_5FdrEhkG7ZE6e8JgHrP1m6zZ90BsJ4jt85FoD23XjfO-8BTOEa5-v3NfnTRMRv-uCfEFJqfJtfBdQy1jhsxwcSQE29JnWIYTgsRdndNCjxZKbVt6fmGSOjhUhHiEEcpB_aES6qVpzEIx25L_Wp1CRu1-DPeBu0XdtkyKuwQUhom8_t-trdTsj-gfzp4m95v8AIj79QXgJnB4ZJjuAM2HlT9DOvZe-8pjrSoFX4ZGmbXPOJLm2Eea27E1UUy_1wPIg7Kns5YqSTASlMNYFNeZVRQFjpD_TNU1ppwpElbxBnkyWBDLuWOAUTUFsDep2Xm-qrdYhIru0sMfo5n144XBvWCNGY9rL4plO2eoawUyWxiSuf8VS1JxLSh-XcI-JAz2z6w=w625-h269-no)

Alejandro Ariza
Thanks And Regards

ShikOfTheRa, 2017-11-15:
That voltage looks dangerously low unless its using a AB7456 or AT7456,

but that said it appears to be showing stats OK as arm / disarm which makes me think its not voltage in this case.

I can't see the version number. 
If it is newer than 3.1.7 Could you see if it works with 3.1.7?

hobbya, 2017-11-16:
Your board uses a MAX7456 and datasheet specifies a minimum voltage of 4.75V. I cannot figure out where you connect the multimeter. Although ShikOfTheRa is suggesting something else, I think it is no harm for you to try powering the osd board with a separate 5V source.

ShikOfTheRa, 2017-11-16:
Agree. If your measuring correctly that is a sign of a problem. In my experience they do not work well if at all at that voltage

ShikOfTheRa, 2017-12-27:
Hopefully OK now. Re-open if still having issues

closed 2017-12-27 by ShikOfTheRa