ShikOfTheRa/scarab-osd #467: GUI won't connect

hventaclu, 2017-11-15:
with a breakout board i was able to flash my OSD board but when trying to use the GUI it just says "connecting" and nothing happens either connected with the same board or with serial pass through. as far as i know, I'm using all the current software and the OSD works correctly, I just can't make any changes with the GUI.
any tips on where to begin troubleshooting? i know nothing about arduino

ShikOfTheRa, 2017-11-15:
Passthrough is notoriously troublesome. If it works, great. if not best to connect direct to OSD
31 Unable to connect with GUI
  * Ensure you have correct baud rate set. Typically 56k for MAvlink/PX4/APM an 115k for all others
  * Try changing baud rate to something else then back again
  * Ensure nothing else connected at the same time
  * Disconnect OSD, reconnect and try closing and restarting APP
  * After launching app try waiting 30 seconds before clicking connect
  * Ensure TX/RX are connected correctly
---

KCrawford100, 2017-11-18:
I am using an CP2102 Breakout and have flashed latest stable MWOSD with no problems to micro and minimum OSDs with arduino with no problems.  I am connected directly to the osd and am having the same problem - message says "connecting".  I have tried setting different baud rates matching port settings as well as setting the config-h to matching baud.  I have tried three different osds that all worked with your earlier versions with the same breakout board. I have tried everything you have suggested above with you luck.  It correctly sees the port but will not directly connect.  I have tried everything for several hours ...
Any other suggestions?  I'm stuck..

hventaclu, 2017-12-19:
ive given up on this board. the osd works so i'm cool for now i just can't make changes to it. i tried connecting directly and different baud rates (everything else i use is 115k). the only i/o i've been successful with is flashing the mwosd software onto it through arduino. i didn't disconnect it from the fc so maybe i'll try that. but for $8 it's not really worth messing with. If it's just a bad board, i'll just buy another one. thanks for the prompt feedback. i just couldn't get around to troubleshooting it as my other quad was requiring more attention.

closed 2017-12-27 by ShikOfTheRa