ShikOfTheRa/scarab-osd #469: GUI Win32 standalone Release 1.7

DerDominik, 2017-12-08:
Hi there,
as I read, the Google Chrome Apps will be closed in future and several other configurators are also released as a Win32 standalone application.
Are there any plans for MWOSD GUI, which were already payed via Google Store?

eephyne, 2017-12-09:
And more of that, what about other operating system. I don't really want to have a virtual machine just for mwosd and I'm not sure it will running flawlessly with wine.

ShikOfTheRa, 2017-12-09:
Google have advised chrome apps will be available on the store until such time they have provided an interface allowing them to run on the desktop. I am not 100% sure what that means or the timing.
Right now there is also a Windows and Mac client available. At some point we will probably have to transition store users over.

closed 2017-12-27 by ShikOfTheRa