ShikOfTheRa/scarab-osd #470: Home Distance Incorrect APM Pixhawk

NorcoT, 2017-12-14:
Hi, first of all great work and thanks for your efforts!

I am running v1.7.3 installed onto micro minimosd using MWOSD GUI Chrome App, it's connected to a mini pixhawk running APM arduplane. All works as it should on the OSD except the distance from home.

From the pixhawk blackbox log one of my flights shows a distance of 2668m from home which also matches up with a meaasure on google maps. DVR of my OSD shows a home distance of 1119m. So the OSD is about a factor of 0.419 out.

I seen that someone else had this problem previously which was apparently solved for v1.7.2? Something to do with APM not scaling?

[image]

[image]

ShikOfTheRa, 2017-12-15:
OK. Will have a look. Any chance of seeing the full flight start to end including initialisation and landing summary?
Send me via pm  on RCG/FPV lab or gary(at)flypix(dot)co(dot)uk if want to keep it private

Thanks

ShikOfTheRa, 2017-12-15:
Off topic, but early next year hope to have more missing PX4 stuff like airspeed / wind direction etc. its in development right now

NorcoT, 2017-12-15:
Yes no problem, I will upload the DVR footage and the blackbox .bin file this weekend and send you a link. 

Another thing I noticed is that my RSSI value did not change, it may be my APM configuration however so I will check first and raise another issue if it isnt the case.

ShikOfTheRa, 2017-12-18:
OK - you have uncovered an interesting issue.... 

at first glance, MWOSD looks like its OK - and maybe its the  method/formula of determining distance is quite inaccurate. It is same formula used in many OSD and other code like multiwii etc. etc

So I will have to look at some other options... .
I will try to simulate in excel first. I need to keep code as simple as possible. But it also needs to be accurate

ShikOfTheRa, 2017-12-19:
So it looked like it was the formula used, but actually I think it was calculating scaling for latitude on GPS home before home was set so effectively getting no scaling. 
At your latitude, scaling is significant. E.g. 1113m without, 2544m with. Sounds in the ballpark for what you are seeing.

So now it calculates continually. No problem with processing power, but does take a few more bytes of memory. 
I’ll have to look into a simulator for mavlink data. But for now I think will need to verify with an actual flight.

BTW, your landing strip and flight area looks awesome….  I drove past there only a couple of months ago on a trip around the Scottish North Coast. Fantastic place
And that battery voltage is interesting. Looks like battery is struggling…

Please can you try 1.7.3.4 and let me know if it solves the issue (will need to tick "Show Test" on firmware flasher)

NorcoT, 2017-12-20:
Thanks for the feedback, is this the portion of code which calculates distance between two sets of lat/long coordinates and takes into account the scale as you approach the poles?

If this is where the issue is i'll have a look into it myself too and let you know if I come up with anything.

Thanks,

Tony

ShikOfTheRa, 2017-12-21:
Yes - I think initial scaling was being taken at lat / lon of 0 - so no scaling... 

This is now updated in main repository and available to test ...
Please can you try 1.7.3.4 and let me know if it solves the issue (will need to tick "Show Test" on firmware flasher)

closed 2017-12-27 by ShikOfTheRa

NorcoT, 2017-12-28:
Hey,

Finally got out to test the new release, its looking spot on! I've attached screenshots of the APM blackbox data at max distance and this ties in with the DVR of the OSD. Also to check the distance matches google maps too.

Also thanks for the update that recalibrates altitude and home distance. This is great as previously all I was getting was altitude above sea level which is great if your flying a real aircraft, not sure useful for FPV..

Thanks for your help, appreciate all your efforts.

Tony

ShikOfTheRa, 2017-12-29:
Thanks for the feedback Tony. Was pretty sure it was resolved but wanted to hear your confirmation.

About the AGL / ASL for altitude, the changes in last release were only for GPSOSD users. I don't think anything changed for APM. In 1.7.2 released end July was a fix. Do you think you would have been using a version prior to that?

I think we can improve that part further - put similar home update o improve accuracy. really pleased how effective it is for GPSOSD users and the next release of this will be very much aimed at improvements for PX4 / APM guys.
[image]

NorcoT, 2017-12-30:
Oh yes, I must have been running 1.7.0, you can see on my original DVR footage from the start of this issue that the home arrow is still the old one.

Thanks again, looking forward to the update in the near future!