ShikOfTheRa/scarab-osd #472: Yaw value instead of throttle

VadimGromov, 2017-12-19:
Compiled with mavlink supports shows the yaw value instead of throttle. Also could not enter osd menu, so may be other channels mismatches too

ShikOfTheRa, 2017-12-20:
Hi,

What FC are you using? 

I have had some challenges in keeping everyone happy with channel assignments as I think for some FC sending over Mavlink they do not use same assignments as PX4/APM. I am wondering if I should add precompiled options for mavlink. 

Compiling with PX4 or APM gives:
  #define ROLLSTICK        1
  #define PITCHSTICK       2
  #define YAWSTICK         4
  #define THROTTLESTICK    3

Compiling with all others gives:
  #define ROLLSTICK        1
  #define PITCHSTICK       2
  #define YAWSTICK         3
  #define THROTTLESTICK    4

Unless override with parameters in:
/********************   RC TX Settings     *********************/

VadimGromov, 2017-12-20:
Thanks. I have changed the defaults  // RX CHANEL IN MwRcData table  #define ROLLSTICK        1  #define PITCHSTICK       2  #define YAWSTICK         4  #define THROTTLESTICK    3and the throttle value works fine. But I could not enter osd menu anyway. I'm using APM with ArduPlane V3.4.0 for a fixed wing. I have inverted Ele in my transmitter. So I tried any position of the pitch/roll stick with with middle/right of another one but there is no result. Have checked in MissionPlanner, values of all four channels are correct.

ShikOfTheRa, 2018-01-10:
OK - in next version, GUI will have an option to select TX type.
It will also be possible to change tx direction in config.h:

//#define TX_P reverse              // Reverse Pitch for OSD controls
//#define TX_R reverse              // Reverse Roll for OSD controls
//#define TX_Y reverse              // Reverse Yaw for OSD controls
//#define TX_T reverse              // Reverse Throttle for OSD controls

closed 2018-01-10 by ShikOfTheRa