ShikOfTheRa/scarab-osd #473: Flash 19200 -> waiting for OSD

unitware, 2018-01-21:
Hi, I'm selecting baud rate 19200 and then firmware flash -> ok
Then I try to connect and all I get is "Waiting for OSD"

Should this work? Or am I misisng something?

Configurator 1.7.1.9 on a mac
Flash settings:  1.7.4.1, minimosd, inav, aero

I'm flashing a micro minimosd

unitware, 2018-01-21:
I compiled my own fw and now I can connect to 19200 from the configurator. But I was expecting an error message if this was not supposed to work in the first place.

ShikOfTheRa, 2018-01-21:
Unfortunately the GUI has no idea of what speed the OSD is running at. It is possible to autodetect, but that process would take some time to run through all the options.
Just how it is...
Glad you got it sorted

unitware, 2018-01-22:
Ok, my usecase was that I have a microminimosd at 115200 that I want to reconfigure to 19200. I was gien the impression that I should set the baudrate in the dropdown and then click flash to flash it with that new setting. 

Is this the way to do it? (I read it somewhere that it could be done like so for 57600 at least)

ShikOfTheRa, 2018-01-22:
The only way is to compile your own firmware. The GUI is intended to cover 90% of requirements. Unusual scenarios like 19.2k 

All the hex files that can flash from GUI re prebuilt for 115k speed expect for mavlink based ones which are 56k. Up until maybe a eyar ago there was no option to flash pre built ones. Every time had to use Arduino to upload!

unitware, 2018-01-22:
ok, I was misled in my assumptions that the 115200/57600 could be extended to 19200 (which is good for softserial on iNav), thanks anyway for the support!

closed 2018-01-22 by unitware